ThunderFly-aerospace/TF-G1 #119: Ohybací přípravek na podvozkové trubky

kaklik, 2018-04-13:
Přípravek na ohýbání trubek bych si představoval asi jako desku opatřenou rádiusy, které je potřeba naohýbat. 

Nejdřív se asi budou ohýbat ohyby 2 a 3.  A až po nich se ohnou konce trubky ohyby 1 a 4. 

S ohyby 1 a 4 bude pravděpodobně největší komplikace, protože je potřeba, aby výsledek byl kolmý na osu ohybů 2 a 3. Je možné, že to bude potřeba zařídit nějakou další deskou, která se přes pant připojí k té první.

kaklik, 2018-04-21:
K radiusům pro ohyby budou potřeba ještě rovné segmenty. Které budou sloužit k přidržení trubky před rádiusem

Elizaa62, 2018-05-02:
Vytvořila jsem díl 888_1011. Zatím je to pokus o radius 2 a to sešikmení trubky pod úhlem. Ale upozorňuji dopředu, že je to jen hrubá verze. Nemám slovosled srovnaný.

kaklik, 2018-05-03:
Myslím, že tak nějak by to být mohlo. :)

kaklik, 2018-05-10:
Prohlížel jsem si aktuální stav a myslím že by mohlo stačit, aby ta trubka při ohybání byla přidržená jenom drážkou do poloviny průměru trubky.  Odstranil by se tím současný problém s modelováním rovných boků, které stejně bude dost problém vytisknout.

Elizaa62, 2018-05-15:
Vždyť to je stejný systém, akorát je to přesnější. Protože ta konstrukce s tím válcem byla taková od oka. Pokud se ti tohle nelíbí, tak to opět předělám do té původní verze.

Elizaa62, 2018-05-16:
Nechala jsem ten přípravek v dílu 888_1011 a vytvořila jsem nový soubor 888_1013. Vím, je to komplikované a asi ti to přijde nepřehledné, ale mě to takhle pomáhá, když je to zvlášť.. Pak to můžu dát časem do stejného souboru, nebo to nějak sjednotit.

V tom novém souboru 888_1013. Základní model je 888_1013. Díl 888_1013_A je taková mnou vymyšlená upínka, která má zobáček. Je zasunuta do dílu 888_1013 a zároveň spojena tím zobáčkem s kvádříkem na dílu 888_1013, co je hned na začátku. Dle mého názoru by to mohlo fungovat jako držák té trubky. Ale nejsem si jistá, zda by se to nevysunulo... Je to jen pokus. Čekám na námitky :) Kdyby ti tenhle způsob přišel dobrý, tak bych to potom ještě rozšířila dál. Jinak ten díl 888_1013_A by se musel tisknout samostatně. 

Pak díl 888_1013_B je vlastně intersection dílu 888_1013 na základní tisknutelnou velikost.

kaklik, 2018-05-16:
Ten přípravek je čím dál tím lepší. Budu k tomu ale mít připomínky, protože mě napadá lepší řešení. :)
Například přesunutím drážky pro trubku zespodu na bok ohybu se dá zlepšit zajištění trubky před ohýbáním. Protože trubka pak může být zajištěna jednoduchým výstupkem ze základny v rovné části bez ohybu.

Elizaa62, 2018-05-17:
Drážky jsem pro trubku jsem přesunula do boku do spodní části tvarovací části. 

Zkusila jsem vymyslet nový systém  upínek. Systém upínání si představuji tak, že až se trubka vloží do drážky, tak do otvoru v základní kostce vloží upínka 888_1013_A, která by tam tu trubku v drážce měla přidržet. Protože si myslím, že když vytvořím pevné výstupky na základní části, tak si nedokážu představit, jak chceš trubku do drážky dostat. 

Ale pokud by ti tenhle opět můj divný inovativní způsob nevyhovoval, tak to předělám :).

kaklik, 2018-05-17:
Myslím že mi to úplně vyhovuje. :)  
Vyndavací upínka je dobrý nápad, protože ta trubka by tam sice šla vložit, protože by se dala v rovném stavu nasunout do drážky. Ale pak by po ohnutí zřejmě už nešla vyndat a to mi úplně nedošlo. :)
Takže tam upínky nech tak jak je máš, jenom je udělej masivnější, protože takhle dimenzované by se při ohýbání asi ulomily. :)

Ještě teď přemýšlím nad tiskem. Nešlo by tu základnu nějak snížit a udělat jí rovnou místo zkosené? Ušetřilo by se tím hodně tiskového času.  Taky by asi bylo dobré, aby v základně byly díry na zapuštěné šrouby, kterým to přišroubuji na nějakou desku.

kaklik, 2018-05-18:
Myslím, že ta část pro ohyby 2 a 3 už bude fungovat :)  
Prohlédl jsem si podrobněji ten díl F pro ohyb 1 a myslím si, že ten by bylo potřeba vylepšit. Protože tak jak je konstruovaný zřejmě nepůjde ohnout trubka do toho žlábku. 

Navrhoval bych pro ohyby 1 a 4 vytvořit separátní přípravek. Do kterého se dá podvozková noha už s vytvořenými ohyby 2 a 3. Pak by se ohyby 1 a 4 vytvořily najednou ohnutím celé trubky. (Doufám, že to půjde, protože ohnout dvě trubka najednou může být dost těžké)

Tím pádem se může podstatně zmenšit tloušťka základny pro přípravek na ohyby 2 a 3. Takže se to bude podstatně lépe a rychleji tisknout. 

Samostatné přípravky pro ohyb 1 a 4 pak mohou být zase šroubovací naležato na desku. Takže taky nebudou muset být tak masivní.  Navíc nepotřebují svorky, protože mohou držet rovnou část trubky proti desce a trubka se prostě zasune mezi ně a desku.  Pak se celá noha ohne nahoru od desky a zas se vyndá.

Elizaa62, 2018-05-21:
Vytvořila jsem nový přípravek pro ohyb 1 a 4. Podle toho, co teď čtu, tak asi zase bude špatně.

kaklik, 2018-05-23:
K tomu, aby to bylo dobře stačí ty ohyby otočit o 90 stupňů a posunout tak, aby trubka byla při ohýbání zapřená o desku. Celé kvůli tomu, aby bylo možné přípravky našroubovat na dřevěnou desku.  A zasunout do nich polotovar nohy s ohyby 2 a 3. Pak se polotovar ručně ohne od desky a tím vzniknou najednou oba ohyby 1 a 4.

Elizaa62, 2018-06-04:
Upravila jsem přípravek pro ohyb 2 a 3 a o 90°jsem otočila přípravek pro ohyb 1 a 4. Přípravky pro ohyb 1 a  4 mají kódy G, H, I, J.

kaklik, 2018-06-07:
Díval jsem se na tohle

[image]

Je to zas trochu lepší.  Akorát ty ohybací přípravky pro ohyby 1 a 4 nebudu mít čím přišroubovat na desku. 
Myslel jsem to tak, že trubka bude před ohybem držená deskou. To znamená, že trubka musí být strčená mezi desku a přípravek a pak se ohne tak, že se celá podvozková noha ohne nahoru směrem od desky.

Elizaa62, 2018-06-13:
Opraveno.

kaklik, 2018-06-14:
Paráda.  To by myslím mohlo fungovat, zkusím to vytisknout a uvidíme. :)

[image]

Elizaa62, 2018-06-26:
Jak to vypadá? :)

kaklik, 2018-07-30:
Zjistil jsem že do ohýbacího přípravku podvozkové nohy nejde zasunout trubka.  Důvod je ten, že drážka pro ohyb obepíná víc než polovinu průměru trubky. 

[image]

Zřejmě se musí zapuštění trubky nějak posunout, aby to tak nebylo.

Elizaa62, 2018-11-20:
Máš nějaké rozměry k novému ohýbání podvozkových trubek?

kaklik, 2018-11-21:
Rozměry bohužel nemáme, protože jsem to řešil hodně operativně a prostě jsem ty trubky ručně ohnul.  Ale myslím si, že délka trubky měřená od ohybu pasujícího do příčné trubky na konec na který je namontovaný kolečko je 180 mm. 

Přesné rozměry by ale měly vypadnout z opravené sestavy  #166 Protože když vírník bude stát na kolečkách, tak horizontální trubka má být vodorovně.

Elizaa62, 2019-02-02:
Mohl bys mi prosím poskytnout nějaký obrázek toho, jak jsou nyní zohýbané ty podvozkové trubky? Abych si mohla udělat nějakou představu o tom, jak mám ten přípravek vyřešit?

kaklik, 2019-02-02:
[image]
[image]

kaklik, 2019-06-12:
Outdated.

closed 2019-06-12 by kaklik