ThunderFly-aerospace/TF-G1 #130: Zkosení ocasních ploch shora a zdola

kaklik, 2018-05-20:
Nyní čtvercové ocasní plochy by bylo potřeba zkosit o 10stupňů jak ze spodní tak i z horní strany. Důvody k tomu jsou následující.

1.  Ušetření hmotnosti 
2. Zjednodušení startu (ocasní plocha nebude při startu tak snadno dřít o zem)
3. Zvětšení prostoru mezi rotorem a ocasní plochou.

Elizaa62, 2018-11-20:
Pokusím se ty plochy zkosit.. ale nedaří se mi tam dát to dno.

Elizaa62, 2018-11-20:
Je to na netu.. nevím , zda jsem to pochopila správně...

kaklik, 2018-11-21:
Dobrá, zatím je to bez dna.  Jak vypadal ten problém s jeho doplněním? 

Jinak to zkosení jsem myslel tak, aby se ten díl zužoval směrem dozadu.  V případě tohoto obrázku nahoru: 

[image]

kaklik, 2019-06-10:
Issue se stalo irelevantní díky možnosti změnit celkovou konstrukci ocasních ploch.

closed 2019-06-10 by kaklik