ThunderFly-aerospace/TF-G1 #139: Zlepšení odolnosti spojky pro vertikální trubku

kaklik, 2018-06-19:
Při experimentech se ukazuje, že pevnost spojení vertikální trubky a horizontální trubky není příliš dobře dimenzována. 
Navrhoval bych proto dva šrouby změnit na M4 a ze spodní strany spojky přidat materiál a šroub M3, aby trubka byla spojkou lépe obepnuta. 

[image]

Elizaa62, 2018-06-26:
Zvětšila jsem spojku pod horizontální trubkou o hodnotu "material_plus". Tato hodnota je uvedena přímo v kódu spojky 666_1017. Označené šrouby jsem změnila i s matkami na velikost M4. V přidaném materiálu jsem vytvořila otvor pro šroub M3.

kaklik, 2018-06-26:
Prohlédl jsem si to a myslím, že to skutečně zlepší pevnost spojky.  
Jenom zvažuji, jestli konstantu material_plus nenahradit spíše tím, že spojka bude tlustší všude.  Například dvojnásobně, než ostatní spojky. 
Bude to mít sice vliv i na otvory v podložce, ale pevnost spojky by se tím zvýšila výrazně. 
Teď je totiž nepříjemné, že vertikální pylon se na podélné trubce poměrně snadno mele. Nevím jak moc to vadí, ale chtěl bych zkusit najít řešení takové, aby se pylon nemlel.

kaklik, 2018-06-27:
Tak jsem přišel na komplikaci, která tedy zřejmě už existuje dlouho, ale teď je výraznější. 
Spojka totiž přesahuje pod spodní rovinu desky. Tím pádem tlačí na dva díly, které jsou určené pro náklad a prohýbá se tím stěna boxu pro náklad.  
Zřejmě je potřeba nákladový prostor nějak tvarově upravit, aby nevznikala kolize s touto spojkou.

kaklik, 2018-06-28:
Zvětšil jsem tu tloušťku spojky tak, aby odpovídala dvojnásobku tloušťky ostatních spojek.  Potřebná délka šroubu pak je 45mm. 
[image]

Nebylo to ale tak úplně jednoduché. Protože nestačilo změnit požadavek na tloušťku v parameters.scad Neboť kopie výpočtu délky šroubu byla pro mne nepochopitelně ještě v souboru samotné spojky. Tyto dva výskyty se pak navzájem přebíjely.  Řešením bylo smazat výpočet vyskytující se přímo u spojky.

 Když jsem se ale pak díval jak je to u ostatních spojek, tak to vypadá, že každá spojka má výpočet ještě u sebe.  Je k tomu nějaký důvod to takhle řešit?

Elizaa62, 2018-09-11:
A jaké výpočty myslíš?  U spojky 666_1017 byl výpočet opravený, jen jsem ho asi zapomněla uvnitř samotného kódu smazat.

kaklik, 2018-09-11:
No jde mi o to, že každá spojka ještě ve svém souboru obsahuje třeba tohle:

https://github.com/Elizaa62/Components/blob/e773c52085e90c9cde41d03ff38ef7d4bda3eefb/src/666_1004.scad#L7-L19

U některých spojek jsem to už smazal a nechal jsem to akorát v souboru s parametry.  Protože když se to neudělá, tak nefunguje změna parametrů v souboru Parameters.scad neboť výpočet u samotné spojky má přednost.

Elizaa62, 2018-09-17:
Opraveno 1004, 1014, 1017, 1026

kaklik, 2018-09-17:
Není, myslím, že je to vyřešené. :)

closed 2018-09-17 by kaklik