ThunderFly-aerospace/TF-G1 #146: Držák pro start ze střechy automobilu

kaklik, 2018-08-21:
Pro start vírníku bude potřeba mechanismus který umožní start ze střechy automobilu.   Pro základní experimenty je nutné, aby konstrukce byla především bezpečná. 
Takže bude realizovaná tak, že na příčníky bude našroubovaná deska z voděodolné překližky o rozměrech 250x125cm tlustá 15mm. 

Na tuto desku bude teprve našroubovaný vypouštěcí mechanismus, tak že v desce budou narážecí matice M10 do dřeva. Do kterých se z druhé strany našroubují šrouby. 
Samotný vypouštěcí mechanismus může být realizovaný z hliníkových profilů a tištěných dílů. 
Odjištění vírníku během vypouštění předpokládám modelářskými servy například [HD LW-20MG](https://hobbyking.com/cz_cz/power-hd-lw-20mg-servo.html) ovládanými ze satelitního modelářského přijímače. 

Uchycení vírníku před opuštěním držáku musí splňovat tyto parametry

- Volnost pohybu v ose pitch alespoň 0 až +20 stupňů
- Volnost pohybu v ose yaw maximálně +/-5 stupňů
- Volnost pohybu v ose Roll +/- 15 stupňů. 

Konvence pojmenování os je patrná z [tohoto obrázku](https://upload.wikimedia.org/wikipedia/commons/c/c1/Yaw_Axis_Corrected.svg).

Postup vypouštění bude následující.

1. Aktivace avionických systému UAV a kontrola vypouštěcího mechanismu. 
2. Postupný rozjed automobilu na max 50km/h. 
3. Růst otáček rotoru
4. Zvednutí přídě UAV na doraz vypouštěcího mechanismu, díky dostatečnému vztlaku rotoru.
5. Autopilot vyrovná náklony v ose Roll. 
6. Odjištění vypouštěcího mechanismu a přechod do stoupání

roman-dvorak, 2018-09-11:
Vypouštěcí plošina by měla mít možnost nějakým způsobem zafixovat vírník ve výchozí poloze. Ať už z důvodu pojezdu auta bez předpokladu startu nebo uvedení vírníku do vodorovné (rovnoběžné s autem) polohy pro usnadnění přizpůsobení autopilota.

Toto by šlo vyřešit tím, že kardanův závěs bude umístěn pevně v jednom místě a okolní "podlaha" (podpěry vírníku) bude výškově nastavitelná. Tím zároveň může dojít k plynulému nastavování rozsahu volnosti ve všech osách.

Při rozjezdu s předpokladem startu by se nejdříve auto rozjelo na určitou rychlost a teprve poté by se podložka pomalu snížia (zvětšila by se volnost kardanova kloubu do uvedených mezí). Rychlost uvolňování by byla podle toho, jak se regulátor dokáže plynule přizpůsobovat.

kaklik, 2018-09-11:
To by ale znamenalo, že kromě ovládání elekromagnetu by se se musela ovládat ještě výška podložky.  Pokud by to skutečně bylo nutné takhle řešit, tak je asi lepší vertikálně posouvat uchycení kardanova závěsu třeba vertikálním trapézovým šroubem na krokovém motoru.

Myslel jsem, že by zafixování pro převoz bylo řešeno uchycením podvozkových ližin do tištěných U profilů našroubovaných na překližku, kde budou i kola.  
Při přípravě pro start by se jenom sundaly upínací pásky a ližiny by ležely v profilech, dokud by nedošlo k nadzvednutí.  

Teď mě napadlo, že to jde vylepšit tím, že by profil pro ližiny měl směrem nahoru rozšiřující se tvar, takže by s roztoucí výškou nadzvednutí stoupala i volnost do stran.

kaklik, 2018-09-11:
Tady je rychlý náčrtek toho, jak bych si to zhruba představoval. 

[image]

Ještě to rozkreslím na dílčí komponenty.

kaklik, 2018-09-17:
Přesunutí příspěvku @roman-dvorak 

Přemýšlel jsem dnska o té Markovo připomínce/otázce, proč je to tlačné a né tažné.

Je pravda, že s tažným závěsem by jsme možná získali podstatně větší stabilitu letadla na závěše, protože by to byla stabilní poloha. Takhle se to vychýlí ze středu a bude v nestabilní poloze. Dokud nebude mít vírník větší tah než rychlost auta.

Zato v případě tažného závěsu by letadlo bylo stále ve stabilním stavu a odepnulo by se v bodě, kdy tah je nulový  a letadlo by okamžitě přidalo na svém výkonu.

Samotný závěs by mohl být tažný, ale umístěný více v zadu než AZ (YAW) točna a tím by zde opadnul problém s tím, že letadlo má vystartovat dopředu. Kotva na letadle by byla umístěna v zadní části podvozku stejným způsobem. Možná by to mělo výhod, že by kontva moha být napevno přidělána a jen by se otáčel magnet na závěsu. Snad je to z následujícího schématu vidět případně překleslím.

[image]
(rozhodně není v měřítku)

díly nakreslené v #156

kaklik, 2018-09-17:
Tohle řešení by zřejmě bylo vhodnější. Protože nedestabilizuje celý systém. Ideální by zřejmě bylo, aby ty rotační osy procházely těžištěm UAV, pak by vliv na změnu stability byl neutrální.

Ten tenzometr bych asi z prvních pokusů vyjmul a nejdřív vyzkoušíme jak se to chová bez něj.  Myslím, že je dost možné, že k ničemu ani nebude potřeba.

kaklik, 2018-09-17:
Ještě je možná zajímavý tenhle obrázek, který ukazuje, jak se měří síly ve skutečném aerodynamickém tunelu.

[image]

roman-dvorak, 2018-09-17:
Myslím, že umístění rotace v ose Roll je špatně ve "třetím dílu". Protože při náklonu v Roll se vírník bude otáčet po velké kružinci okolo třetího dílu. Což mi nepřipadá dobře. Tato rotační osa by měla být taktéž co nejblíže těšišti a ose Roll na vírníku.

A zde by mohl vzniknout "čtvrtý díl", který by řešil rotaci ve dvou osách zároveň. Rotaci v ose Roll tak, že by přímo do magnetu byl M8 inbus šroub. V tom samém dílu by byla volnost v ose Pitch (Podruhé!) aby zde nedocházelo ke kroucení mezi magnetem a kotvou.

kaklik, 2018-09-17:
S tou vzdáleností od osy rotace maš úplně pravdu.  Jenom si teď nedovedu moc představit ten nový čtvrtý díl. 
Dovedl by jsi ten díl nakreslit? 

S tím jestli  se pod virník vejdeme bych si zatím nedělal moc starosti.  Vírník můžeme nad desku zvednout tím, že pod ližiny dáme nějaké podložky. Zkomplikuje to akorát montáž rotoru. Nemusí být navíc nutně uchycený za ližiny ale můžeme klidně použit tu vertikální trubku, pro třetí kolečko, která je teď nevyužitá.

roman-dvorak, 2018-09-17:
"čtvrtý díl" bude nějaká spojka, pomocí které se spojí 2 M8 závitové tyče. Pro vodorovnou tyč bych navrhoval ukončit uvnitř dílu. Do dílo by byla kapsa pro M8 matici a z venku by se to stáhlo proti této matici. Svislá závitová tyč by byla skrz snažená z vnější stran maticí. Toto řešení navrhuji udělat z důvodu, aby výška tohoto zařízení byla co nejmenší. 

[image]
(Na obrázku je to nakresleno přesně obráceně) 

"Pátý díl" se nasadí na M8 závitovou tyč. Uvnitř tohoto dílu bude kapsa pro matici a stáhne se to oproti druhé matici z venku. V horní části tohoto dílu bude díra pro osu ALT (pitch).

"Šestý díl" bude pomocí takově vydlice nasazen na "pátý díl". Ložiska jsou uložena z vnější strany tohoto dílu. V horní části budou z obou staran znovu vložena ložiska a pomocí šroubu M8 to bude zašroubováno přímo do těla magnetu. Na horní straně budou takové dva "výběžky", které budou nastavovat rozsah rotační osy tohoto kloubu. Skrz tyto výběžky budou zašroubovány nastavovací šrouby skrz matice zasunuté do kapes v tomto díle.

[image]

Mezi "Šestým dílem" a magnetem bude "Sedmý díl", který bude mít stejný průměr jako magnet a na sobě takový "trn" který toto omezí. Trn může být vystužen zašroubováním šroubu do těla trnu (ze strany od magnetu) Okolo díry pro M8 šrob by bylo vhodné udělat osazení, které by tlačilo na střední část ložiska.

[image]

kaklik, 2018-09-29:
Aktuální chyby v návrhu: 

- Díl 888_3007 má špatnou velikost matek pro připevnění příruby EMFS-17.

kaklik, 2018-10-03:
Tak velikost děr pro matky jsem vyřešil. 
A nový způsob upevnění k magnetu také. 
Takže zbývá jenom nalézt metodu, jak počítat tvar závěsu pro zvětšení rozsahu úhlů. Na tom pracuje @roman-dvorak.

kaklik, 2018-10-03:
Aktuálně držák na autě vypadá takto: 

[image]

Elizaa62, 2018-11-20:
Jak to vypadá s tímto issue?

kaklik, 2019-02-04:
Tenhle úkol se zatím stabilizoval na použití pevného sloupku s elektromegnetem. 
Budoucí problém je změna konstrukčního uspořádání tak, aby držák pro start umožnil i umístění nákladu na UAV.

closed 2019-02-04 by kaklik