ThunderFly-aerospace/TF-G1 #150: Díl pro přidělání koleček k hliníkové tyči udělat bytelnější

roman-dvorak, 2018-08-28:
Díl č. 666_1014 [https://github.com/Elizaa62/Components/blob/master/src/666_1014.scad](url) pro přidělánů koleček k AL tyči upravit tak, aby byl k tyči přidělán pomocí dvou M3 šroubů. Nyní se díra v hliníkové trubce vykloktá a kolečko nedrží kolmo.

kaklik, 2018-08-30:
Ono to možná na těch vyrobených kusech je díky tomu, že vyvrtat ty díry úplně kolmo je dost těžké.  Nicméně použít dva šrouby je dobrý nápad.

Elizaa62, 2018-09-11:
Do dílu 666_1014 přidán druhý šroub pro připevnění k trubce.

kaklik, 2018-09-11:
@Elizaa62 Přidání šroubu je myslím správně, je fajn, že se ty šrouby vešly do stejné šířky.  Ještě je to potřeba upravit u vrtacího přípravku, ale to předpokládám víš.

Elizaa62, 2018-09-11:
Vrtací přípravek je také opraven. Jen jsem to sem nezapsala.

kaklik, 2018-09-11:
Já, když jsem se sem díval, tak v repozitáři ještě nebyl. Teď už ho vidím. :)

closed 2018-09-11 by kaklik