ThunderFly-aerospace/TF-G1 #153: Přípravek pro zkracování šroubů

kaklik, 2018-09-09:
Na zkrácení šroubů do kování rotorových listů by se hodil přípravek, do kterého se zašroubuje šroub (proti matici na opačné straně přípravku) a pak se pilkou na železo prořízne štěrbina v definované vzdálenosti od hlavy šroubu. 
Tím by mělo být možné opakovaně řezat šrouby stejné délky.

Podobné řešení už existuje pro řezání trubek. viz https://github.com/kaklik/Components/blob/master/src/manufacturing_helpers/tube_cutter.scad

Elizaa62, 2018-09-11:
Je uveden jako "screw_cutter".

kaklik, 2018-09-13:
Tak jsem trochu upravil rozměry a vytiskl to. 
[image]

Myslím, že to funguje :) 

[image]

Akorát asi bude potřeba šroub do přípravku uchytit samojistnou maticí, protože se při řezání povoluje.

closed 2018-09-13 by kaklik