ThunderFly-aerospace/TF-G1 #156: Kardanův závěs pro uchycení elektropermanentního magnetu

kaklik, 2018-09-11:
Při uchycení podvozku vírníku na desku na střešním nosiči musí být umožněna jistá volnost pohybu minimálně ve dvou rotačních osách.  Viz #146 
K tomu je potřeba užít kardanův závěs s omezeným rozsahem pohybu.  Rozsah pohybu může být omezen tvarem. 
Jako čepy pro spojení jednotlivých dílů bych navrhoval použít šrouby M6  nějakých standardních délek s inbus hlavou.

Magnet, který máme  je typ EPPM-6329-24-020 s průměrem 63mm. 
https://www.magnety.cz/pridrzne-elektromagnety/pridrzne-elektromagnety-s-permanentnim-magnetem/

kaklik, 2018-09-12:
Zkusil jsem udělat náčrtek, jak by konstrukce závěsu mohla vypadat.  
Pro spojení předpokládám použití šroubů M6 a samojistných matic zapuštěných v kapsách. Šrouby se do matic nebudou úplně dotahovat a tím vznikne možnost protáčení v kloubových spojích.

[image]

[image]

roman-dvorak, 2018-09-16:
### První díl (AZ, YAW) bude řešit rotaci kolmo na střechu auta (YAW osu auta). 

tento díl by se pomocí 2-4 M6 šroubů přidělal do dřevěné desky (18mm). Skrz tento díl povede licovaný šroub 8mm M6, který bude zespoud (od dřevěné desky přidělán bezpečnostní maticí). Pod maticí bude ložisko 608. Stejné ložisko bude i shora, kde z horní strany pnese druhý díl.

[image]

### Druhý díl (ALT) 

Druhý díl bude přidělán na první díl pomocí stejného lícovaného šroubu. V této části bude hlava šroubu a povede směrem dolů. Možná by stálo za to, místo, kde bude hlava šroubu udělat širší pro možnost podložky pro roznesení síly.

Z tohoto dílu, směrem dozadu, povede "výběžek", ve kterém bude zašroubován šroub (M6?) směrem dolů. Bude vytvářet takový "trn" pro omezení volnosti rotace v AZ ose.

[image]

Z boků do tohoto dílu povede další lícovaný šroub, který zajistí rotaci v ose ALT.

### Třetí díl.
Tento díl bude jednoduchá kostka, kterou skrz na skrz povede lícovaný šroub z druhého dílu. Ve spodní části ním bude prostrčena M8 závitová tyč. Tato tyč bude uložena rotačně pomocí radiálních ložísek 608 z boku dílu. Ložiska budou z každé strany stažěna maticí M8.

[image]

Na konci závitové tyče bude umístěn tenzometr a magnet. Je vybrána závitová tyč M8, která už je dostatečně pevná a navíc v magnetu je M8 závit. Na druhé straně musí být závaží, které bude vyrovnávan nevyváženost soustavy.

### Celek...
[image]

kaklik, 2018-09-17:
Souhlasím, jenom bych kvůli konstrukčnímu zjednodušení navrhoval díl První díl (AZ, YAW) tvarově upravit tak, aby vycházel z kuželu. 

[image]

Zapsáno jako 

    cylinder(h=50, r1=50, r2=30, center=false);

Odpadne tím požadavek na ta žebra a celkově se to myslím zjednoduší i pro tisk.

Ostatně u druhého dílu (ALT) by to mohlo být podobné. 

[Rozměry ložiska 608](https://www.arkov.cz/p/608-2zr-lozisko-kulickove-jednorade-oboustranne-kryte-plechem-zkl-13967)

kaklik, 2018-09-17:
Tohle jsou lícované šrouby průměru 8mm/M6, které máme  k dispozici
Délky jsou 70 a 30mm.  Pro ukázku je na fotce ještě ložisko 608 a čep 8mm / 40mm, který ale do ložiska nejde prostrčit.

[image]

kaklik, 2018-09-18:
"čtvrtý díl" bude nějaká spojka, pomocí které se spojí 2 M8 závitové tyče. Pro vodorovnou tyč bych navrhoval ukončit uvnitř dílu. Do dílo by byla kapsa pro M8 matici a z venku by se to stáhlo proti této matici. Svislá závitová tyč by byla skrz snažená z vnější stran maticí. Toto řešení navrhuji udělat z důvodu, aby výška tohoto zařízení byla co nejmenší. 

[image]
(Na obrázku je to nakresleno přesně obráceně) 

"Pátý díl" se nasadí na M8 závitovou tyč. Uvnitř tohoto dílu bude kapsa pro matici a stáhne se to oproti druhé matici z venku. V horní části tohoto dílu bude díra pro osu ALT (pitch).

"Šestý díl" bude pomocí takově vydlice nasazen na "pátý díl". Ložiska jsou uložena z vnější strany tohoto dílu. V horní části budou z obou staran znovu vložena ložiska a pomocí šroubu M8 to bude zašroubováno přímo do těla magnetu. Na horní straně budou takové dva "výběžky", které budou nastavovat rozsah rotační osy tohoto kloubu. Skrz tyto výběžky budou zašroubovány nastavovací šrouby skrz matice zasunuté do kapes v tomto díle.

[image]

Mezi "Šestým dílem" a magnetem bude "Sedmý díl", který bude mít stejný průměr jako magnet a na sobě takový "trn" který toto omezí. Trn může být vystužen zašroubováním šroubu do těla trnu (ze strany od magnetu) Okolo díry pro M8 šrob by bylo vhodné udělat osazení, které by tlačilo na střední část ložiska.

[image]

kaklik, 2018-09-18:
@Elizaa62  Správně jsou ty nákresy od @roman-dvorak

Elizaa62, 2018-09-18:
Díl první jako díl 888_3000. 
Díl druhý jako díl 888_3001.

roman-dvorak, 2018-09-18:
Na díl 888_3000 je potřeba přidělat časti pro omezení rotace v YAW ose.

@Elizaa62 Po domluvě s @kaklik by to chtělo na díl 888_3000 přidělat takové "sloupky", vyšší než je současná výška dílu. Na mém nákresu výše je to trochu vidět. Skrze tento díl umístit sroub, kterým by se nastavoval rozsah volnosti osy a tento šroub by narážel do výběžku z dílu 888_3001. Matice by byla umístěná v kapse přímo v tom sloupku.

Z výběžku na dílu 888_3001 by se mohla odstranit díra, která byla původně myšlená na trn směrem dolú, který by měl stejný účel.

roman-dvorak, 2018-09-19:
@Elizaa62 "Čtvrtý díl" a "pátý díl" by bylo možné spojit v jeden. Viz. následující nákres
[image]

Udělal bych to spíše tak, že "díl 4+5" by byl jako vidlice a "šestý díl" by se vkládal dovnitř. Ostatní díly by měly zůstat stejné.

Elizaa62, 2018-09-19:
Doplněny sloupky k dílu 888_3000 i s otvorem pro šroub a kapsou pro matku. Akorát si nejsem jistá, zda správně.

Zvětšeny rozměry dílu 888_3001 a odstraněn otvor pro šroub ve výstupku.

Nový díl 888_3002 -  třetí díl z tohoto issue.

Nový díl 888_3003 - sestava předchozích třech dílů.

roman-dvorak, 2018-09-19:
@Elizaa62 úplně správně ne :)

Má to být tak, že díl 888_3001 se má na dílu 888_3000 otáčet v omezeném rozsahu (cca 10deg) A šrouby vložené do těch výstupků z dílu 888_3000 mají nastavovat rozsah, jak moc se může otáčet. 

Takže je potřeba přidělat kapsu i do druhého sloupku. 
Výstupek z dílu 888_3001 bych udělal trochu uzší a odstranit díru skrz.
Sloupky z dílu 888_3000 udělat dál od sebe a udělat větší mezeru mezi sloupkem z dílu 888_3000 a válcovou základnou dílu 888_3001

Nápad od @kaklik, že by to bylo bez žeber a byl by to kužel není špatný, protože by se tím velmi zvýšila pevnost těchto výstupků.

kaklik, 2018-09-19:
Nezavedeme si nějakou konvenci v těch názvech?  vytvořit sestavu z dílů je dobrý nápad, ale chtělo by to pojmenovat nějak aby bylo jasné, že to sestava té skupiny. 

Zatím  to máme odlišené jako 

- 666_ výkresy od pana Dobiáše. 
- 888_ naše nové výkresy

A pak je zevedeno číslování skupiny jako třeba  888_3XXX jsou díly závěsu.  Tohle mi připadá dobré. 
Akorát bych to ještě rozšířil o sestavy a zavedl třeba 888_31XX konkrétně třeba 888_3101. Za předpokladu, že skupina dílů nebude mít víc jak 100 komponent. Pak by to asi přestalo fungovat. 

Jde mi o to, že když se sestavy budou číslovat stejně jako díly, tak bude existovat sestava 888_3003 a další díl skupiny bude 888_3004. Což mi přijde dost nepřehledné.

Elizaa62, 2018-09-20:
Sestava přejmenována na 888_3101.scad

888_3003 - čtvrtý a pátý díl
888_3004 - šestý díl
888_3005 - sedmý díl

roman-dvorak, 2018-09-20:
[image]

Díl 3004 by měl být dole půlkruhový, aby se mohl v dílu 3003 otáčet. To že nepůjde naklonit dopředu dolů nevadí. Musí jít naklánět "dozadu".

Díl 3004 musí být širší a zarážky pro omezení roll rotace dále od sebe. Aby se díl 3005 mohl otáčet alespoň o 10deg na každou stranu. Ze stejného důvodu musí být díl 3004 nižší.

roman-dvorak, 2018-09-20:
@Elizaa62 Když si prohlížím takhle ten složený díl, tak vzhledem k rozměrům a potřebné volnosti roll osy (minulý komentář) by možná bylo lepší vidlici udělat na díl 3004 a díl 3005 do toho vkládat. Omlouvám, se že jsem to navrhoval původně obráceně. Nedokázal jsem si dobře představit rozměry.

To zapuštění ložiska do vnitřní strany vidlice je fajn nápad. Ale blbé je, že když se šroub stáhne ze stran maticí, tak se nebude otáčet a nemá tam to ložisko moc smysl. Proto by bylo dobré to udělat podobně jako je to na mé modifikací dílu 3002. A vlastně ten díl 3003 lze vytvořit modifikací dílu 3002 tak, že se přidá uložení pro ložisko i do druhé díry a trochu se protáhne.

Elizaa62, 2018-09-20:
Pokusila jsem se to opravit.

kaklik, 2018-09-21:
Díly 3003 a 3004 do sebe pasují velmi na těsno. není tam žádná vůle pro separační podložky u ložisek.

[image]

Tohle se netiskne moc dobře. 
[image]

Ve spodním dílu není žádná rezerva na délce šroubu a šroub sotva dosahuje do spodního ložiska 
[image]

kaklik, 2018-09-22:
Díry pro hlavy šroubů M6 v základně by měly být spíše šestihranné,  protože takto jdou použít pouze šrouby se šestihrannou hlavou.

kaklik, 2018-09-29:
V dílu 888_2006 chybí kapsy na zapuštění matic šrobů M2.5 od serva.

kaklik, 2019-02-04:
Pro současnou aplikaci řešení vyhovuje.

closed 2019-02-04 by kaklik