ThunderFly-aerospace/TF-G1 #158: Náhrada řemenového převodu hlavy ozubeným

kaklik, 2018-09-18:
Na hlavě by bylo potřeba vyzkoušet nahradit současný řemenový převod ozubeným převodem. 
Důvody k tomu jsou:

- Zvýšení převodového poměru mezi motorem a rotorem 
- Odstranění zátěže rotoru vznikající statickým odporem při ohybu řemenu

Ozubená kola by zřejmě měla mít šípové ozubení.  První verzi ozubených kol je potřeba navrhnout tak, aby je bylo možné vytisknout.

kaklik, 2019-10-15:
Ozubený převod vyžaduje změnu smyslu otáčení rotoru.

closed 2019-10-15 by kaklik

roman-dvorak, 2019-10-15:
> Ozubený převod vyžaduje změnu smyslu otáčení rotoru.

Asi nechápu... Vždyť se motor může táčet obráceně. Nevidím zde v tomto žadný problém.

Ale toto řešení nepovažuji za vhodné z důvodu pružení desky rotorové hlavy.

kaklik, 2019-10-16:
Když se motor bude otáčet opačně, tak se povolí unašeč pastorku na jeho hřídeli, pokud by neměl levý závit. 
Musel by se tak změnit smysl otáčení rotoru, ale to je velká konstrukční změna.