ThunderFly-aerospace/TF-G1 #164: Tisknutelný díl pro montáž rotorové hlavy na pilon

kaklik, 2018-09-25:
Plechové díly 666 1021 a 666 1022 mají být nahrazené jedním tisknutlým dílem s kloubovým ložiskem [EFSM-12 ](https://www.igus.cz/product/382).

roman-dvorak, 2018-09-25:
[image]

kaklik, 2018-09-26:
Prohlédl jsem si to a myslím že ten díl je už z velké části hotový. Našel jsem zatím jenom tyhle nedostatky

- chybí díry na šrouby od serva, 
- drážka na kolík z desky hlavy je otočená o 90 stupňů
- prostřední díra pro kulové ložisko je malá. Měla by mít průměr 25mm. 
- servo by mělo být blíže k pravému okraji dílu vychází mi to podle výkresu asi na 4mm od hrany plechu.
- stěna pod ložiskem by myslím měla být tlustší, bude se přes ní přenášet celá hmotnost.

kaklik, 2018-09-26:
Ještě jsem si všiml, že ty rozměry, které tam jsou zřejmě nejsou pro přírubu EFSM-12 ale pro menší EFSM-04. Na tom odkazu je totiž potřeba si překliknout na správný typ. Potom jsou rozměry následující: 

[image]

Parameter | Value
-- | --
Ø d1 | 12,00 mm
Ø dB | 25,0 mm
h | 40,0 mm
m | 28,0 mm
A1 | 7,0 mm
Ag | 13,0 mm
N | 5,3 mm

kaklik, 2018-09-27:
Tak díl jsem vytiskl.  Pozitivní je, že do něj pasuje příruba s kulovým kloubem. :) 
Jsou tam ale i nedokonalosti. 

[image]

Například tyto: 
- Nejde prostrčit servo do otvoru, protože tomu brání jeho kabel. Řešením je zvětšit otvor na kabely ve směru osy Y. 
- Ovory pro upevnění serv mají špatnou rozteč ve směru osy Y.

kaklik, 2018-10-03:
Dobře, po téhle úpravě by to pak mělo být stejně široké, jako původní díl. To by mělo být správně. :)

U toho čepu jde o to, že při pohybu ve směru osy X naráží do okrajů toho otvoru. Nevím o kolik to je, ale zřejmě je potřeba jednak zvětšit ten otvor, a potom i zvětšit poloměr podstavy toho kuželu, kterým je otvor vytvořen.  To by mělo zajistit, aby se čep mohl pohybovat ve větším rozsahu úhlů a po větší vzdálenosti. 
Rozsah úhlů ve kterých se čep může pohybovat je asi +/- 20 stupňů.

kaklik, 2018-10-19:
Zjistil jsem, že by bylo zřejmě lepší tenhle díl upravit tak, aby se na trubku šrouboval fixní délkou šroubů.  S tím že hlavička a matka šroubu bude v dílu trochu zapuštěná. Vhodný šroub je například M4x50, neboť je potíž sehnat delší šrouby.

closed 2018-12-29 by Elizaa62