ThunderFly-aerospace/TF-G1 #166: Aktualizace sestavy podle současného stavu

kaklik, 2018-10-02:
Soubory ve složce https://github.com/Elizaa62/Components/tree/master/src/drawings 

Už neobsahují přiliš aktuální stav celkové sestavy.  Bylo by proto zřejmě vhodné sestavu aktualizovat.

Elizaa62, 2018-12-29:
Sestava je aktualizovaná pod názvem 666_2000.  Sestavu pod názvem 888_1005 jsem smazala, aby nedošlo k záměnám.

Elizaa62, 2018-12-29:
> Sestava je aktualizovaná pod názvem 666_2000. Sestavu pod názvem 888_1005 jsem smazala, aby nedošlo k záměnám.

kaklik, 2019-02-04:
Rozměry trubek jsem přidal přímo jako fotografie do úlohy řešení ohýbacího přípravku #119

kaklik, 2019-05-31:
- V modelu chybí pohyblivé části směrovky
- Některé díly třeba kryt rotorové hlavy není upraven, aby pasoval k současnému stavu modelu

Elizaa62, 2019-06-10:
Koukala jsem, že díl A spodního krytu je opravená. Opravila jsem podle rozměrů podvozkové trubky. 

Mohl bys mi prosím napsat seznam dílů, které se měnily a tudíž nepasují do sestavy?

Kryt rotorové hlavy je který díl?

kaklik, 2019-06-10:
Dílů, které do sestavy nepasují díky tomu, že se měnily je minimum. 
Je to v podstatě akorát:
-  [Kryt vertikální trubky](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/666_1032.stl)
-  Navázání držáku motoru k hornímu krytu

Horší je to ale s díly které v sestavě chybí úplně.  Ideálně by sestava totiž měla obsahovat všechny tyto komponenty: 

- vrtule
- rotor
- sestava rotorové hlavy
- okrytování těla vírníku (možná s krytem motoru)
- ocas včetně pohyblivých částí
- podvozek přední
- podvozek zadní
- kolečka včetně středu

rotorové listy, závaží a kování máme namodelované jako jednotlivé komponenty, je z toho však potřeba vytvořit podsestavu.

Elizaa62, 2019-06-10:
A k těm chybějícím dílům máš když tak jejich čísla? Abych to nemusela hledat a zkoumat.

kaklik, 2019-06-10:
Aktualizoval jsem ten můj předchozí komentář o odkazy na díly, které teď aktuálně máme.  
Ještě mě napadlo, že by ti možná ušetřilo práci a čas, kdyby jsi tyto díly do té sestavy[ importovala jako už vyrenderované STL soubory](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/STL_Import_and_Export#import). Pak bys nemusela čekat, až se vyrenderuje celá sestava. 

Tento proces bude minimálně nutný v případě rotorového domku, protože ten nemáme jako openscad soubor, ale pouze jako STEP model, který ti překonvertuju na STL přidám mezi komponenty.

Elizaa62, 2019-06-12:
Vytvořila jsem sestavu rotorových listů s podložkami a kováním, ale bez závěsu. Nejsem si jistá, jak to tam vše patří.

kaklik, 2019-06-12:
Ta hláška je způsobená tím, že někdo udělal změny v repozitáři mezitím, co jsi něco upravovala. Nemělo by to být nic hrozného a  mělo by stačit spustit git pull. 

Teď se mi taky podařilo exportovat [díl rotorového domku](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/666_1207.stl).

kaklik, 2019-06-13:
Vytvořil jsem ti [příklad](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/src/666_1207.scad) jak importovat už hotové díly do sestavy.

Tenhle nový způsob považuji za důležitý kvůli tomu, že to je megarychlá metoda, jak skládat sestavy z vyrenderovaných dílů. Kde nemusíš čekat, až se ti překreslí sestava.

kaklik, 2019-06-17:
Ta vrtule v té sestavě je, akorát má špatnou velikost a je na nečekaném místě. 

[image]

kaklik, 2019-06-17:
Tak problém s vrtulí jsem částečně vyřešil tak, že jsem model vrtule v meshlabu přeškáloval na reálnou velikost v milimetrech.

roman-dvorak, 2019-06-18:
dílčí sestavy byly složeny v blenderu. Výsledek je takový:

[image]

rotorové listy byly do modelu vloženy ručně, protože je nebylo možné vyrenderovat v sestavě.
Místo prerotátoru byl přidán válec.

kaklik, 2019-06-18:
@Elizaa62  Aby se tohle issue mohlo uzavřít a příště nebylo potřeba tu sestavu skládat v blenderu, tak je potřeba vyřešit ty nezaškrtnuté body výše. 

Měly by ti k tomu chybět akorát rozměry válce jako hlavního motoru, které jsou průměr 49.5 mm a výška 49 mm .

kaklik, 2019-08-07:
Myslím, že tohle je pro aktuální stav vírníku vyřešeno.

closed 2019-08-07 by kaklik