ThunderFly-aerospace/TF-G1 #168: Výška vrtacích přípravků

Elizaa62, 2018-10-03:
Je nutné zkontrolovat výšku vrtacích přípravků.

Elizaa62, 2018-10-03:
[image]

Toto je můj nákres, když jsem tvořila vrtací přípravky.

Elizaa62, 2018-10-03:
Přípravky jsem překontrolovala a nezdá se mi,že by tam byla nějaká chyba. 

Vložila jsem přípravky pro znázornění do souboru drill_helper_together. Tento soubor se pak může klidně smazat.

kaklik, 2018-10-03:
No ono se nám minule, když jsme vrtali tu hlavní trubku podařilo vyvrtat díry přesně tím kolmým způsobem, jak je tam máš nakreslené, protože ty přípravky k sobě tak pasovaly. :)

Jinak od toho souboru drill_helper_together.scad už existuje delší dobu lepší varianta, která vypadá takhle: 
[image]

a je to soubor https://github.com/Elizaa62/Components/blob/master/src/drawings/888_1007.scad  
Asi by se ho hodilo přejmenovat na  drill_helper_together.scad, protože je to výztižnější název a pravděpodobně ho tak spíš někdo najde.

kaklik, 2019-08-14:
Většina vrtacích přípravků se nyní změní, takže jejich přesné parametry nyní není potřeba řešit.

closed 2019-08-14 by kaklik