ThunderFly-aerospace/TF-G1 #169: Zkusit vytvořit tisknutelnou variantu desky rotorové hlavy

kaklik, 2018-10-04:
Frézovanou desku rotorové hlavy by bylo zajímavé vyměnit za tisknutelný díl později s nějakými plechovými vložkami.

[image]

kaklik, 2018-12-04:
A tohle je plechová náhrada současné  frézované desky, která tam teď je namontovaná: 

[image]

kaklik, 2019-01-15:
Otvory pro očka na nové desce je potřeba posunout dál od řemenice, aby očka nepřekážela řemenici.

kaklik, 2019-02-06:
Kapsy pro matky šroubů táhel by zřejmě měly být na opačnou stranu, aby při demontáži šroubu matka nevypadla.  Nebo alternativně z bogu ze zadní strany, kde bude otvor i lépe chráněn před zátékáním vody.

kaklik, 2019-06-10:
Současná tisknutelná rotorová deska funguje velmi dobře, ale jen v malých úhlech, kdy hrany tisknutelných dílů nenarazí při zpětném chodu do sebe.  Díl je potřeba upravit tak, aby hrany dílů nemohly přesahovat přes sebe a nebo měly šikmý náběh, který zabrání zaseknutí.

roman-dvorak, 2019-06-29:
Díl se nyní otáčí volně v obou dvou osách. Problém je dost vysoká Yaw volnost. Jako jedno z možných řešení je celý tvar kloubu výrazně protáhnout. Následně vytisknout celý a otestovat.

kaklik, 2019-08-08:
Nahrazeno #225

closed 2019-08-08 by kaklik