ThunderFly-aerospace/TF-G1 #170: Vytvoření pohledů vhodných pro příručku

kaklik, 2018-10-26:
V příručce budou potřeba pohledy s okótovanými základními rozměry.  V podstatě to vychází z úkolu #166 

Asi by to mělo být vyřešené tak, že každý obrázek v příručce bude mít tady svůj vlastní soubor: https://github.com/Elizaa62/Components/tree/master/src/drawings
Který bude využívat celkovou sestavu.  Sestava je teď v souboru  	888_1005.scad nebo 666_2000.scad netuším, jaký je mezi nimi rozdíl. 

Obrázky by měly být rozděleny na svoje soubory, aby je bylo možné v příručce jednoduše aktualizovat. Ideální by bylo, kdyby sobor pohledu měl i nastavený pohled přes $vpr, $vpt a $vpd aby se dal vždycky snadno přegenerovat.

kaklik, 2019-03-08:
Jeden z potřebných pohledů by měl být takový, aby umožňoval zanesení těchto kót (se správnými rozměry):

[image]
 
Je to důležité ke správnému nastavení simulačního modelu.

Elizaa62, 2019-05-05:
Co vše na tom pohledu má být vidět?

kaklik, 2019-05-05:
Ano, nemusí tam být vůbec ty vnitří části, ani tam nemusí vůbec být rotor a kolečka.  Stačí pouze okótované jejich pravděpodobné pozice.

closed 2019-10-01 by kaklik