ThunderFly-aerospace/TF-G1 #172: Úprava držáku motoru 666_1026

kaklik, 2018-11-14:
Protože tisknutý díl 888_1015 už úplně nahradil spojku 666_1026 na rotorové hlavě, tak už není potřeba aby spojka 666_1026 zachovávala svůj tvar společný pro obě umístění.  Je tak možné zjednodušit uchycení motoru.  Tím že se ze spojky vytvaruje díl specifický pro motor. 

Původní idea uchycení motoru plechem byla takováto: 

[image]

Nejproblematičtější částí jsou ohyby plechu, které je těžké vyrobit přesně. 

Do zakótovaných otvorů měly patřit šrouby dílu 666_1026. Nyní by ale díl 666_1026 bylo možné tvarově upravit tak, aby plech pro uchycení motoru nebyl potřeba.  Zároveň není nutné mít nastavitelný náklon seřízení motoru protaženým otvorem pro šroub. Neboť úhel může být nastaven přímo parametrem vytištěného dílu. 

Potenciální komplikace úpravy je  snížení  možnosti chlazení motoru odvodem tepla do plechového dílu. To ale lze vyřešit zachováním plochého plechu pouze v přední části bez ohybů.  Následující úkoly proto jsou: 

- [ ] Upravit tvar dílu  888_1026, tak aby obsahoval i boční stěny v původním návrhu tvořené plechem. V těchto bočních stěnách bude zapuštění pro matky.
- [ ] Vytvořit nákres předního plechu tak, aby plech bylo možné přišroubovat z přední strany k dílu 888_1026, za kapsy pro matky umístěné v bočních stěnách dílu 888_1026

roman-dvorak, 2019-10-06:
Motorový díl pro nový vírník může být o něco nižžší a zároveň může mít menší sklon osy motoru.

kaklik, 2019-10-07:
Tohle issue je k dílu, který už neexistuje. Vytvořil jsem pro to nové issue #252.

closed 2019-10-07 by kaklik