ThunderFly-aerospace/TF-G1 #177: Držák na střechu auta pro transport s jedonu osou volnosti

roman-dvorak, 2019-01-28:
Pro potřeby vývoje SW a nastavování regulátoru by bylo potřeba vyrobit držák, který bude umožňovat pohyb pouze v ose Roll. Osy Pitch a Yaw by měly být pevně zafixovány. Držák by měl vírník podpírat pod ocasní částí pomocí "dvojnožky". Držák by měl jít snadno demontovat nebo nějak sklopit.

Tento držák je rozšíření issue #146 

Na následujících obrázcích je můj návrh tohoto držáku.
[image]

Spodní díly 1-1 a 1-2: Spodní díl bude přivrtán a sešroubován do nosné desky auta. K dílu 1-1 bude pomocí šroubu přidělán díl 1-2. Spolu budou tvořit klou, který celý držák umožní sklopit dozadu. Do dílu 1-2 budou nastrčeny AL trubky a sešroubovány pomocí šroubu skrz.
[image]

Díl 2-1 a 2-2 Pevně spojí trubky od patek. Trubky zde budou znovu prošroubované.
[image]

Vrchní díl umožňí připevnit horizontální trubku vírníku. Je potřeba mít možnost naklonit tento díl podle sklonu vírníku.
[image]

roman-dvorak, 2019-01-29:
Výška středu horizontální trubky vírníku nad základní deskou je 28 cm. V první řadě je potřeba vytvořit držák, který bude v roll ose volný. Tzn. ocasní trubka se v držáku musí volně protáčet.

kaklik, 2019-01-30:
Nějakým speciálním protáčením bych to teď nekomplikoval. Myslím že stačí když se na tom dílu, co stahuje trubku úplně nedotáhnou šrouby.

Elizaa62, 2019-02-02:
V tomto issue jsem vytvořila díl 1.1 a 1.2 jako díl 888_3011.scad. Bylo by možné vytvořit pull_request, abych si mohla stáhnout i další vytvořené díly?

kaklik, 2019-02-02:
pull-request obsahující nový držák jsem vytvořil.

kaklik, 2019-02-04:
Tak se po posledním testu ukázalo, že držák je potřeba mírně snížit.  Ovšem do desky na střeše TF-R1 jsme navrtali díry pro současnou rozteč patek držáku. 

Rozteč by měla být patrná z následujících fotografií a je to potřeba přepočítat na rozteč děr v patkách. 

[image]

[image]

Aby nebylo potřeba vrtat nové díry, tak je nutné tuto současnou vyvrtanou rozteč děr zachovat. 
Držák pak lze snížit buď 

- snížením výšky patek, což jde, neboť teď je zapuštění pro šrouby k desce zbytečně vysoké a stačila by výška pouze na matku M6. 
- Změnou délky trubek, což ale přináší i nutnost zmenit úhel trubek v držáku. Trubky jsou teď dlouhé 199mm.

Elizaa62, 2019-02-07:
díl 888_3015.scad

Elizaa62, 2019-02-08:
Patky opraveny.

kaklik, 2019-02-11:
Držák funguje.

closed 2019-02-11 by kaklik