ThunderFly-aerospace/TF-G1 #180: Otočit držák pro GPS anténu

roman-dvorak, 2019-02-10:
Držák z issue #25 je potřeba otočit tak, aby byla orientace podle následujícího obrázku. Trojúhelník by měl mít jednu těžnici rovnoběžně s osou vírníku. 

[image]

Protože by jeden šroub vycházel přímo do horizontální trubky vírníku, tak to lze udělat tak, že podpírající sloupek bude o trochu širší a vyšší (je možné, že současné rozměry sloupku jsou dostačující). A na jeho spodní straně bude kapsa pro zasunutí matky.

[image]

kaklik, 2019-02-10:
A není lepší spíš přijít na to, jak softwarově odrotovat ten senzor? 
Protože vyměnit celou podložku je poměrně drastický servisní zásah.  Navíc ještě k tomu současný držák docela funguje.

roman-dvorak, 2019-02-12:
Ano, už asi vím, jak to otočit softwarově. Tohle issue nemusí být tedy vyřešeno hned. Ale pro další kus vírníku by to tak mohlo být. Navíc, když by z nějakého důvodu bylo potřeba, tak toto řešení umožňuje větší variabilitu umístění GPS na podložce.

kaklik, 2019-02-13:
Pokud je známé softwarové řešení problému. Tak bych to navrhoval neřešit v modelu. Neboť další kus stejně bude mít GPS modul, který bude vypadat jinak.

roman-dvorak, 2019-07-06:
Na novém kusu vírníku bude použita jiná GPS. Sloupky pro GPS budou dostraněny

Vyřešeno pomocí #212 a #214

closed 2019-07-06 by roman-dvorak