ThunderFly-aerospace/TF-G1 #181: Nové uchycení akumulátorů

kaklik, 2019-02-15:
Vzhledem k velikosti nových akumulátorů, je potřeba vymyslet uchycení mimo hlavní trup. 

Jednou z možností jsou boční závěsy, které by držely dvě symetrické gondoly. Gondoly by měly mít nějaký přibližně aerodynamický tvar, pro minimalizaci odporu. Akumulátory si lze pro účel návrhu gondoly představit jako kvádr univerzálních rozměrů a nezanedbatelné hmotnosti.   (Přibližné parametry jsou rozměry 12x8x20cm a hmotnost například 3kg)

Uchycení pak musí splňovat tyto požadavky:

- Variabilitu pro různé typy akumulátorů
- Parametrizaci pro změnu polohy těžiště celého stroje
- Možnost odnímatelnosti za účelem dobíjení

Elizaa62, 2019-03-05:
Měl bys prosím nějaký návrh toho, jak by to zhruba mělo vypadat? Nedokážu si to představit....

kaklik, 2019-03-06:
Zatím ještě konkrétní představu nemám. Před začátkem vytváření modelu je zřejmě potřeba udělat nějaké nákresy tužkou na papír. Zatím jsem ani nepředpokládal, že by to podle tohoto šlo nakreslit. 
Zkusím udělat nějaký nákres.