ThunderFly-aerospace/TF-G1 #182: Úprava držáku kotvy magnetu

kaklik, 2019-02-25:
Současný držák kotvy je potřeba upravit tak, aby k němu bylo možné přišroubovat novou kotvu. Jejíchž výkres vypadá takto: 

[image]

Elizaa62, 2019-05-05:
Upravila jsem průměr z 70 na 75. Dále rozteč šroubů jsem nastavila dle uvedeného výkresu na 42,5. Oboje jsem raději nastavila parametricky. 
Je zde ještě něco dalšího, co je potřeba udělat?

kaklik, 2019-05-05:
Super, tohle bych řekl, že je tím hotové. Jenom to teď vytisknu a vyzkouším, jestli to pasuje. :)

closed 2019-05-05 by kaklik

kaklik, 2019-05-07:
Pasuje to! :) 

[image]