ThunderFly-aerospace/TF-G1 #184: Nový testovací přípravek na měření charakteristik rotoru

kaklik, 2019-02-25:
Pro měření rotoru by bylo potřeba vytvořit nové díly vycházející ze současné konstrukce rotorové hlavy. 
Tak aby bylo umožněno rotor naklonit pod definovaným úhlem dozadu a měřit jeho vztlak a odpor. Náklon rotoru do stran není třeba uvažovat. 
Důležitá je fixace náklonu v nastavitelném úhlu v rozsahu 0° (vodorovně) až -30° (sklon dozadu).

Elizaa62, 2019-05-05:
Víš trochu přesněji, jak si toto představuješ?

kaklik, 2019-05-05:
Bylo několik variant na kterých jsem se chtěl domluvit. 
Nyní to vypadá, že konstrukční řešení se už ustálilo na nějakém [takovémto provedení](https://github.com/ThunderFly-aerospace/Auto-G2/issues/10).

Což znamená, že díl který bys měla nakreslit bude vypadat přibližně nějak takto:

 
[image]

Jeho smyslem je, že bude držet rotor a snímač otáček. Celý se přišroubuje čtyřmi šrouby M4 k horní části [tohoto dílu](https://github.com/ThunderFly-aerospace/TF-R1/blob/master/CAD/STL/888_5008.stl). (Z důvodu tisku je díl otočený vzhůru nohama, takže montážní otvory jsou dole).
@cesilko1  pro montážní díl rotoru připravil [knihovní prvek](https://github.com/ThunderFly-aerospace/TF-R1/blob/master/CAD/src/lib/rotor_joint.scad), který  by mohl konstrukci dílu usnadnit. 

Somotný nový díl může vycházet z [existujícího dílu rotorové hlavy](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/888_1020.stl). Ze kterého je potřeba odebrat části, které nejsou potřeba. (Zapuštění pro kulové ložisko a výstupy pro páky serv)
Naopak je potřeba na díl přidat z knihovny otvory pro montážní šrouby k [naklápěcímu dílu](https://github.com/ThunderFly-aerospace/TF-R1/blob/master/CAD/STL/888_5008.stl) měřícího přípravku.
Taktéž je potřeba ze spodní strany připravit otvor pro šestihrannou matici M6 hlavního šroubu rotoru, který přitahuje ložiska, jak je nakresleno v náčrtku výše.

kaklik, 2019-08-06:
Tohle je vyřešeno [tímto dílem](https://github.com/ThunderFly-aerospace/force_balance/blob/master/cad/STL/888_5012.stl)

closed 2019-08-06 by kaklik