ThunderFly-aerospace/TF-G1 #187: Vypouštěcí zařízení pro start z lana

roman-dvorak, 2019-06-03:
Do dílu 888_1026 (Držák tažného motoru) je potřeba integrovat vypouštěcí zařízení, které umožní odpojit tažné lano pomocí dálkového ovladače a serva na vírníku.

Princip zařízení je vidět na tomto orázku:
http://www.gliders.uk.com/images/0723470.jpg

V dílu 888_1026 bude podélná štěrbina. U vstupu bude vysoká, aby umožňovala velký úhel tažného lana, a její výška se bude postupně snižovat. Příčně, přes tuto díru, bude osa - dlouhý M3-A2 šroub. Tento šroub je potřeba orientovat tak, aby nedocházelo k jeho povolení tím, že se o něj bude lano smýkat.. O tuto osu se z horní strany bude opírat ocelový drát veducí až k servu, které bude za drát tahat. Po vytažení ocelového drátu servem dojde k uvolnění lana. 
[image]

Navrhovaná realizace je na následujících náčrtcích:

[image]

[image]

Předpokládané umístění serva je takto:
[image]
Servo je potřeba, vzhledem ke směru tisku, orientovat tímto směrem.

Je potřeba brát v ohled, že se skrze tento díl se přenáší veškerý tah motoru a je tedy potřeba co nejvíce zachovat jeho pevnosti.

kaklik, 2019-06-28:
Nepodařilo se mi přijít na to jak je ta konstrukce vymyšlená.  Jedinné umístění serva ve kterém tam pasuje je toto: 

[image]

[image]

Tohle řešení ale má několik problémů. 

1. Páka serva nevychází naproti drážce pro páku a táhlo. 
2. Servo drží pouze na jednom šroubu
3. Přesahuje mimo model do zadní části

Při konstrukci dílu je třeba také myslet na to, že na díl má pak nějak navazovat [horní kryt](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/666_1025.stl).

roman-dvorak, 2019-07-07:
@cesilko1 Myslíš, že takto je to hotové? Mě přijde ta drážka pro příchytku serva dost úzká. Jak je v modelu široká? Jsou tam započitané dostatečné tolerance?

cesilko1, 2019-07-07:
@roman-dvorak jestli jsem správně pochopil tohle http://file.yizimg.com/357982/201719-11844730.pdf tak je ta drážka o 0.5mm větší

kaklik, 2019-07-09:
Mě přijde že tam je hrozně málo místa pro páku serva a taky na úchyt táhla na páku serva. 

[image]

roman-dvorak, 2019-07-11:
Vidím další změny.. 
https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/888_1026.stl

Takhle se mi to už líbí podstatně více. @kaklik co si o tom myslíš teď?

kaklik, 2019-07-17:
Ten model je původně plánovaný pro tisk v [této pozici](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/print_plate_1026_engine.stl).

Nevím jak se vytiskne ten převislý držák serva. Nechal jsem to tisknout s podporami a uvidíme, jaký bude výsledek.

roman-dvorak, 2019-07-20:
Pro tisk bez podpor je potřeba do dílu namodelovat podpory ručně v místě serva a vytvořit drážku pro vyvedení kabelu serva.

[image]

kaklik, 2019-07-22:
Takhle konstruovaná podpora integrovaná do modelu nebude dobře fungovat. 

[image]

Je potřeba aby byla co nejtenčí. Ideálně jen velmi málo nad 0.61mm. Zároveň je také potřeba aby podpírala převis v celém profilu a nebo, aby umožnila jednoznačným způsobem tisknout most.  Vzhledem k tomu, že je převis ukosený, tak to v tomhle případě jednoznačné není. Podpora by tak ukosená musela být taky.

cesilko1, 2019-07-23:
tak jsem upravil supporty, tloušťka se dá nastavit proměnnou `supports_thickness` na řádku 33

kaklik, 2019-07-30:
Rozteč šroubů pro uchycení serva je příliš velká. 
V modelu jsou zřejmě použity šroby M2,5 ale reálně servo má být přichyceno šrouby M2.

commit 8e5d3071 (2019-06-10): fix #187 vypouštěcí zařízení
closed 2019-07-30 by kaklik