ThunderFly-aerospace/TF-G1 #188: Zlepšení odpružení zadních kol

kaklik, 2019-06-05:
Zadní kola jsou nyní odpružena deformací AL trubky, řešení funguje dobře, ale má tyto nedokonalosti. 

- Deformace trubky není definovaná a může snadno dojít k nevratné deformaci, která je těžko rozpoznatelná. 
- Konstrukce se špatně vyrábí, neboť je těžké správně navrtat všechny díry v trubkách

Řešením těchto problémů by mohlo být vytvoření tisknutelné vložky, která deformaci normalizuje na provedení torzní tyče tlumené třením o plastové pouzdro. 

Konstrukčně to lze realizovat 3D tisknutelnou vložkou v trubce. Vložka může vypadat například takto:

(pohled v řezu tlustou podvozkovou trubkou)
[image]

(Pohled z boku)

[image]

Principem nového konstrukčního řešení je, že tenčí podvozková trubka, ke které je uchyceno kolečko je šroubem skrz díru přichycena na konci dlouhého plastového pouzdra. Celé pouzdro je pak vloženo do větší trubky.  Pozice pouzdra je pak zafixována šrouby stahovatelným límcem, který zafixuje jak natočení tak i nasunutí pouzdra na větší trubce. 

Při funkci se pak deformuje krutem trobka o menším průměru uvnitř pouzdra.  Plastové pouzdro zároveň třením tlumí případné kmitání podvozku.

RomanDCZ, 2019-06-06:
Podle mého názoru je nezávislé odpružení zadních kol jednou z příčin nestabilní jízdy. Myslím, že náprava by měla být tuhá, odpružená společně. Nezávislé odpružení kol mění geometrii podvozku. Změna geometrie způsobuje změnu směru jízdy.
Ideový návrh je v příloze. Spojovací tyč musí být tuhá, odpružení může být provedeno pomocí pružných elementů v uložení spojovací tyče nebo pomocí pružin. Tlumení klidně pomocí třecího pouzdra. Z hlediska stability by měla kola mít co největší průměr ( moment setrvačnosti) a podvozek co největší rozvor a rozchod.
Těžiště letounu by mělo v půdorysu být nad těžištěm trojúhelníku tvořeného koly.
Přední kolo může opatřeno natáčením.

[podvozek_navrh.pdf](https://github.com/ThunderFly-aerospace/TF-G1/files/3263361/podvozek_navrh.pdf)

kaklik, 2019-06-06:
Tento návrh jsem nyní konzultoval s kamarádem a nejsem si jistý správností této úvahy.  Neboť za normálních okolností je nezávislost odpružení jednotlivých kol žádoucí, kvůli tomu že není chtěný stav, aby jeden hrbol vytvořil nežádoucí impulz v celé nápravě. 

Při pružení nápravy skutečně dojde ke změně geometrie, ale tato změna je spíše ovlivněna konstrukcí nápravy, než nezávislostí odpružení. 
Ohledne reakce na jizdní stavy by byla lepsi šikmá osu kyvu, aby kola šla do priklonu a použít nějakou sbihavost, ta stabilizuje. 

Pokud ve staticke poloze jsou kola priklonena k sobe se šikmou osou kyvani, tak pri propružení jde naprava proti nezadoucimu odkloneni kola ven a stabilita je jina.

Rozdíl konstrukčního řešení je zobrazen na následujícím obrázku (vrchní obrázek je momentálně implementován). 

[image]

S úvahou ohledně těžiště ale úplně souhlasím.

kaklik, 2019-06-18:
Z hlediska přistání je zřejmě lepší, aby podvozek byl přetlumený a absorboval energii na dlouhé dráze. 
Příkladem takového podvozku může být [G-Force Landing Gear](https://www.youtube.com/watch?v=ajM3yOojPbg)
Tato konstrukce zřejmě ale nebude naopak vhodná pro start.

roman-dvorak, 2019-07-01:
Nový návrh podvozku by měl mít každou polovinu samostatně uchycenou za kostru vírníku a nezávisle a tlumenou. Tento návrh podvozku by měl zamezit, aby se kolečko nemohlo ohnout dozadu a tlumící pohyb byl prováděn pouze nahoru a dolu s tím, že při větším stlačení se kolečko bude naklánět okolo  roll (podélné) osy vírníku.

V tomto issue je celkový nákres podvozku. Jednotlivé díly budou řešeny v samostatných issues, aby bylo možné řešit jejich detaily samostatně a aby na tom mohlo snáze pracovat více lidí. 

[image]

Podvozek se bude skládat ze dvou tyčí, které budou pevně připevněné k vidlici kolečka. Směrem k tělu vírníku se budou rozcházet. Tam budou zavěšeny za ložisko, které se bude otáčet v podélné ose vírníku. V této implementaci bychom použili Igus ložiska [Igubal stojaté ložisko KSTM](https://www.igus.cz/info/spherical-bearings-pillow-block-kst). 

Mezi těmito tyčemi bude umístěn píst, který se bude stlačovat oproti existující podvozkové příčné tyči. Tlumič bude muset být připevněn na obou stranách pomocí kulových ložisek. Buď znovu  pomocí [KSTM](https://www.igus.cz/info/spherical-bearings-pillow-block-kst) nebo pomocí jiného Igubal ložiska. Např. [EFOM](https://www.igus.cz/info/spherical-bearings-flange-bearings-overview). Výběr ložiska bude záležet na konkrétním dílu. Tak, aby tam bylo co nejméně zbytečného materiálu.

K podvozku bude muset být definováno několik základních parametrů, podle kterých se budou počítat požadované úhly a podobné další rozměry.

- Výška podvozku - průnět vzdálenosti podélné tyče vírníku a osa koleček do osy Z
- Rozchod podvozku - vzdálenost koleček
- Vzdálenost připevňovacích míst k podélné trubce vírníku
- Vzdálenost připevnění pístů k příčné trubce

roman-dvorak, 2019-07-01:
Zadním podvozkem by mělo být možné zatáčet. Zatáčení bude řešené tak, že ke každému měchovému kolu #191 bude přes třecí převod připevňen motor s rotačním pláštěm. Tento motor bude sloužit jako brzda. Autopilot bude vždy jedno kolečko přibržďovat a tím bude celý vírník zatáčet. Toto řešení má několik výhod. Například to, že při startu i při přistání jsou zadní kolečka nejdéle v kontaktu se zemí.

K brždění by mělo být možné použít VESC regulátor.

RomanDCZ, 2019-07-02:
Při propružení dojde ke změně geometrie, letoun zatočí. To by musela být
doplněna ještě horní ramena.
Pak nevím, proč u pružiny musí být kulové ložiska. Tam bude stačit obyčejný
čep na každé straně.
Největším problémem bude naladění tuhosti pružiny a tlumiče.

roman-dvorak, 2019-07-02:
Ano, všude by stačilo použít klasická radiální kuličková ložiska. Například ložiska z řady 608. 

Ale chceme použít plastová kulová ložiska Igubal, které vydrží opravdu hodně a nevadí jim nárazy. To by mohl být v případě kuličkových ložisek problém. 

Navíc tato ložiska už mají docela šikovnou přírubu, kterou lze jen přišroubovat k tištěným dílům. Tím se ušetří obrovské množství materiálu. Tím je celá konstrukce i lehčí. 

Tato ložiska budou použita i u zavěšení ramen podvozku, budou se otáčet kolem jedné (společné) osy.

V obrázku je schématicky nakreslena pružina. Na vírníku to bude pneumatický tlumič, který bude mít propojené obě části přes jednostranný ventil.

kaklik, 2019-07-02:
Nerozumím úplně tomu, proč by změna geometrie vedla při této konstrukci k zatočení.   Při propružení dojde na A-ramenu k příklonu kola, což působí naopak [stabilizujícím](https://en.wikipedia.org/wiki/Camber_angle) účinkem a zvyšuje sílu potřebnou k zatočení.

kaklik, 2019-08-11:
Myslím, že konstrukční řešení zadního podvozku je již zvoleno. Nyní je potřeba dořešit realizační detaily.

closed 2019-08-11 by kaklik