ThunderFly-aerospace/TF-G1 #189: Zlepšení odpružení předního kola

kaklik, 2019-06-05:
Upevnění předního kola by mělo být přesunuto do dílu [držáku motoru](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/888_1026.stl). 

Zároveň je potřeba aby bod otáčení kola byl přímo pod místem uchycení na trubu, nebo více vzadu, aby nedocházelo k vyvrácení kola při nárazu na překážku.  Nejpoužívanější řešení jsou tato: 

[image]

kaklik, 2019-08-11:
@roman-dvorak  čím se vůbec liší ty dva obrázky vpravo?  V tom nákresu zřejmě asi chybí parametry popsané v tomto článku

roman-dvorak, 2019-08-12:
Tam se to lišio umístěním kloubů. U jedné konstrukce je píst stále svisle dulů (pod konstatním úlhem) a klouby jsou pouze na 'torque linku'. V této konstrukci jsou také úhly kolečka a závleku konstantní po celou dobu chodu pístu. 

Druhá kontrukce je udělaná tak, že se tyto úhly mění s dle stlačení pístu. V posledním obrázku zřejmě nejsou úplně korektně značené klouby.

roman-dvorak, 2019-08-19:
K absorci nárazu byly zvoleny písty [RT/57208/M/25](https://www.tme.eu/cz/details/rt_57208_m_25/pracovni-valce/norgren-herion/). Použije se jejich dvojic, která bude umístěna ze strany kolečka. Přední podvozek se bude skládat z několika dílů, ktré zajistí 

[image]

[image]

Kolečko bude využito z issue #191. To bude umístěno v H dilu. Z jedné strany do něj bude vlozeno kolečko a z druhé bude pomocí ložisek (696) připvněn druhý díl. Díl se může na horní straně sbíhat. To se uvidí podle poměru velikostí. 

Další díl je taková "páka" která bude pod nosníkem vírníku umístěna na šikmo. To má za úkol stabilizovat pohyb a samotné kolečko. 

V obou dílech musí být přípevňovací body pro připevnění pístu. Dole to bude realizováno pomocí kulového očka IGUS, kdy tam bude muset být díla na M3 pevnostní šroub.  Nahoře bude na píst nasazený vytištěný díl, který bude s pístem sešroubován pomocí M3 šroubu a skrz vidlici povede jeden (asi) M6 šroub. Ten bude uložen ve dvojici ložisek 696. 

Kdyby byly další nejasnosti @cesilko1, tak určitě piš.

kaklik, 2019-08-19:
Ten šikmý díl pro uchycení kola by zřejmě neměl přesahovat dopředu více, než je vzdálenost od místa uchycení k vrtuli. 

Otočná a pevná část by také měla obsahovat nějaké dorazy, aby rozsah otočení směru kola byl omezený.

roman-dvorak, 2019-09-01:
U tohoto dílu je potřeba zúžit díly tak, aby bylo možné použít do ložisek M4 šrouby. Nejdéle tedy M4x55 ale i užší díly by byly dostatečné. 

- [ ] Zúžit ramena
- [ ] Zvětšít caster úhel
- [ ] Prodloužít díl pod podložkou
- [ ] Posunout kolečko více dopředu, počítat to z nastavení úhlu motoru
- [ ] Ve vidlici použít M4 šroub

roman-dvorak, 2019-09-22:
- [ ] Špatně vypočítaná délka osy kolečka
- [ ] Ve středním dílu není otvor na ložiska
- [ ] Chybi zapuštění pro matku a hlavu šroubu ve spodním díle

1) nejdelší M4 šroub, co máme je M4x55. Aktuální rozměry zapuštění šroubu by byly akorát pro M4x60. 
2) V prostředním dílu předního kolečka chybí otvoy pro ložisko
3) Na horní části spodního dílu chybí zapuštění pro matku a hlavu šroubu (osy tohoto kloubu) Díky tomu nelze tento spoj dotáhnout bez použití více nástrojů

closed 2019-09-30 by kaklik