ThunderFly-aerospace/TF-G1 #191: 3D tištěný náboj mechového kola

kaklik, 2019-06-10:
Do mechového kola je potřeba navrhnout 3D tišténé disky do kterých se vloží ložiska s nábojem. 

[image]

Konstrukce disků může být například takováto: 

[image]

Ložiska použitá na kole budou 608. Středový šroub by měl být alespoň M5.  Kolo by mělo umožnit použití jak pro stranovou montáž, tak i pro použití ve vidlici. 

Model je jeden z potřebných pro vyřešení issue #166

roman-dvorak, 2019-06-11:
[image]

budou to dva disky, které budou mít na sobě válec, který je bude držet v nějaké určité vzdálenosti. Budou staženy trojicí šroubů. 

Vnitřní díl se musí dotýkat pouze vnitřní části ložiska. Je vhodné tam udělat takový kružek, který zajistí dostatečnou vzdálenost od zbytku ložiska.

alcamrkvicka, 2019-06-12:
Disky pro mechove kolo ... 888_1029
[image]
[image]

roman-dvorak, 2019-06-12:
tu vnitřní díru je potřeba výrazně rozšířit. Nastavit jí průměr něco jako ```608_bearing_outer_diameter - 5```. Tím se zajistí, aby nedocházelo ke zbytečnému tření mezi vnitřní (statickou) částí ložíska a vytištěným dílem (který se bude točit). 

Dále je potřeba přes tuto díru udělat přemostění, které zajistí lepší kvalitu výtisku. Takto by se při výrobě materiál propadal.

alcamrkvicka, 2019-06-15:
Lozisko cele zapusteno do disku.
[image]
[image]

roman-dvorak, 2019-06-15:
Model obsahuje několik chyb:

- [ ] Disky mají mezi sebou velkou mezeru - zřejmě je to způsobeno špatným použitím parametru
- [ ] Zvětšit průměr vnějšího disku na 51 mm (parametr už taktéž existuje)
- [ ] Zajistit použití šroubu M3x30 - je potřeba správně napočítat jeho délku a správně zapustit jeho hlavu do dílu.

alcamrkvicka, 2019-06-21:
Chyby opraveny, zapusteni hlavy je udelano pro sroub M3x35.
[image]
[image]

kaklik, 2019-06-24:
Tak jsem vytištěný náboj zkoušel složit a mám tyhle poznatky

- [ ] Překrytí otvoru vrstvou materiálu se blbě vyřezává, protože zde není jen jedna vrstva, jejíž tloušťka by byla definována parametrem [layer_thickness](https://github.com/ThunderFly-aerospace/Auto-G2/blob/dd94fedd44c93dcea9dbe41fe11ccf57465f813d/CAD/parameters.scad#L4)
- [ ] Nevychází délka vnitřní vložky pro šroub M5 a kus přečnívá i na složeném nedotáhnutém kole.
- [ ] Středová vložka se blbě tiskne, díky tenké stěně okolo šroubu. S ohledem na použití původně navržené vidlice bych asi stejnak  byl pro změnu šroubu na M4. Myslím že je pro plánovanou aplikaci dostatečný. 

[image]

alcamrkvicka, 2019-06-25:
Melo by to byt spravene. Ta spatna delka vnitrni vlozky vznikla kvuli zmene zapusteni loziska - puvodne melo byt zapustene jen z poloviny.
[image]
Nyni by mezi vnitrnimi vlozkami mela byt 2 mm mezera.

kaklik, 2019-06-25:
Ok, ale mezi vnitřními vložkami mezera být nemůže.  Neboť principem té vložky je právě vlastnost, že ten šroub který vložkami prochází utáhnu tolik, aby se vložky stáhly k sobě. Zároveň ale tahle vzdálenost vyjde přesně taková, že stažením vložek nevznikne boční tlak na vnitřní kroužky ložisek. 

Výsledek má být, že kolo se může volně otáčet na dovou stažených vložkách, ale nemůže dojít k jeho posuvu po vložce.

alcamrkvicka, 2019-06-25:
dobre, je to prodlouzene akorat, aby se dotykaly. Mne bylo na zacatku receno, ze se dotykat nemusi, proto jsem to tak udelala

roman-dvorak, 2019-06-29:
Vidlice ke kolečku je moc úzká - zřejmě nepočítá s vnějšími podložkami. Lépe je to vidět na obrázku
[image]

roman-dvorak, 2019-07-05:
Z vložky do ložiska je potřeba odebrat vnější límec, který měl původně sloužit k ochraně ložiska. To zatím není potřeba a nyní to způsobuje, že vidlice musí být velmi široká.

roman-dvorak, 2019-07-07:
Vnitrnimu dilu odebran vnejsi limec na ochranu loziska pred pocasim. 
Upraveny rozměry náboje, změněno zapuštění šroubu, přidána plná vrstva nad otvor pro matice.

kaklik, 2019-07-13:
Vnitřnímu dílu ložiska chybí osazení, které by se opíralo o obvod vnitřního kroužku a bránilo kontaktu límce s otáčejícím se krytováním ložiska. 
Absence tohoto osazení, které v modelu původně bylo zároveň ovlivňuje boční vůli kolečka na vnitřních vložkách šroubu M4.  Kolečko se nyní může zhruba o milimetr hýbat ze strany na stranu.

kaklik, 2019-07-30:
Myslím že vnitřní kroužek na model @roman-dvorak  doplnil.

closed 2019-07-30 by kaklik