ThunderFly-aerospace/TF-G1 #192: Nový držák UAV na protikus magnetu

ChroustJan, 2019-06-11:
Je potřeba vytvořit nový díl, který bude držet kovovou část, která se chytá na magnet na střeše auta. Je potřeba, aby se aktuální držák posunul o svojí celou délku dozadu. Viz znázornění na fotce. 

[image]

Díl by měl vzniknout úpravou držáku předního kolečka.

roman-dvorak, 2019-06-11:
[image]

Při vytváření tohoto dílu lze vycházet z 888_2003, který byl umístěn na stejném místě. Popřípadě ho rovnou upravit. 
[image]

ChroustJan, 2019-06-12:
Je důležité, aby ten držák byl trochu vysunutý, aby nedošlo ke kolizi s tím modrým dílem držící motor. Podle mě bude určitě stačit 1cm pod tu fialovou desku UAV. Možná i 5mm.

kaklik, 2019-06-24:
Současný držák 888_2003 je vysunutý příliš a zbytečně zvedá vírník nad magnet.

[image]

ChroustJan, 2019-07-02:
Je potřeba provést opravu dílu. Skrz celý materiál je šestihranný otvor. Matice propadá.

[image]

kaklik, 2019-07-13:
Na opraveném dílu kotva magnetu drží.

commit a985725c (2019-06-13): FIX #192

Magnet musel být trochu vyosen oproti původnímu držáku, kvůli udržení způsobu montáže šroubů a jejich rozteče.
closed 2019-07-13 by kaklik