ThunderFly-aerospace/TF-G1 #194: Úprava ocasních ploch

roman-dvorak, 2019-06-15:
Existující řešení ocasních ploch ve tvaru písmene H má řadu problémů, které by vyřešily ocasní plochy ve tvaru písmene A. Pro toto issue lze vycházet z existujícího modelu ocasních ploch. Toto řešení by mělo zvětšit pevnost ocasu a snížit možnost střetu s rotorem.

Nově bude ocas přidělán pomocí dvojice trubek (D=10mm), které budou vycházet z podložky vírníku. 

Servo bude nahrazeno servem s patkami určenými pro montáž na plocho. Tato úprava má hlavní přínos v tom, že bude podstatně snazší servo demontovat a vyměnit. 

Ocasní plochy by se měly skádat ze dvou částí, které budou na horní straně vzájemně sešroubované.

[image]

kaklik, 2019-06-18:
Issue, které se zabývá uchycením serv v ocasních plochách je zde: https://github.com/ThunderFly-aerospace/TF-G1/issues/120

alcamrkvicka, 2019-06-24:
[image]
[image]

Cast pro zapusteni je navrzena na sroub M3 ISO7380

roman-dvorak, 2019-08-13:
Protože pro přidělání ocasních ploch budou použity uhlíkové trubky s vnějším průměrem 10 mm a musí být skrze ně provlečeny alespoň 3 dráty k servům, tak není možné trubku provrtat a sešroubovat. Při tomto postupu navíc hrozí výrazné oslabení pevnosti trubky a její následné zničení vlhkostí. 

Proto byl vymyšlen postup, že na konci trubky bude takový vytištěný "adaptér", který se na trubku nalepí a sešroubuje se zbytkem směrovky. 

Je potřeba aby tento adaptér odpovídal profilu křídla. Ze zadní strany by měl mít otvor na provlečení kabelů. Jeho spojení se zbytkem směrovky by bylo provedeno sešroubováním. Jako nejlepší místo na sešroubování mi připadá umístit šrouby pod a nad připevňovací trubku, kdy by se šroub zastrčil ze předu. V hlavním díle směrovky by byly díry na kapsy pro matky a díry pro šrouby.

Trochu načrtnuté je to zde. Pokud jsou nějaké nejasnosti, ptej se na cokoliv :

[image]

kaklik, 2019-08-13:
Já bych to asi raději viděl tak, aby se ten díl šrouboval ke směrovce zboku, nikoli zepředu. Část toho dílu by tak stále byla permenentně součástí směrovky. Domnívám se, že se to tak bude i lépe tisknout. 

V jaké poloze se vůbec bude tisknout tahle nová směrovka?   Moje představa je, že bude spodní hranou profilu stát na podložce.  
Aby ale v takovém případě mohl být vytištěn otvor pro pohyblivou část směrovky, tak směrovka musí být zkosená.

roman-dvorak, 2019-08-19:
Já jsem totiž myslel, že se to bude tisknout náběžnou hranou dolů. Je blbé, že pak ta hrana není hezká. Aby se to tisklo na stoajto, tak ta zkosená hrana bude muset být dost zkosená. To mi skolo vychází tam vymodelovat podpory ručně a nahoře udělat most. Ta podpora by měla stále tvar airfolu té směrovky jen by tam bylo dutá komora. 

A z toho předpokladu, že se to bude tisknout na náběžnou hranu, tak jsem zvolil šrouby takto umístěné. Při bočním umístění šroubů se bojím toho, aby kvůli šroubům se směrovka zbytečně neztlusťovala.

kaklik, 2019-08-19:
Ten tisk na náběžné hraně není optimální. Vyžaduje v modelu vymodelovaný brim, který je pak třeba odřezávat. 
Použití vymodelovaných podpor místo úkosu je dobrý nápad. 

Na šrouby tam může být boule, která může být i součástí boule pro trubku.

taury55, 2019-08-23:
> 
> 
> Existující řešení ocasních ploch ve tvaru písmene H má řadu problémů, které by vyřešily ocasní plochy ve tvaru písmene A. Pro toto issue lze vycházet z existujícího modelu ocasních ploch. Toto řešení by mělo zvětšit pevnost ocasu a snížit možnost střetu s rotorem.
> 
> Nově bude ocas přidělán pomocí dvojice trubek (D=10mm), které budou vycházet z podložky vírníku.
> 
> Servo bude nahrazeno servem (BMS-115HV) s otvory pro boční přidělání. Tato úprava má hlavní přínos v tom, že bude podstatně snazší servo demontovat a vyměnit. Rozměry tohoto serva mám oměřené.
> 
> Ocasní plochy by se měly skádat ze dvou částí, které budou na horní straně vzájemně sešroubované.
> 
> Tvar ploch by měl vycházet z [existujícího modelu](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/src/666_1028.scad).
> 
> [image]

Ty trubky, na kterých budou plochy uchecny mají jaký rozestup?

roman-dvorak, 2019-09-01:
V modelu je několik nových problémů:

 * Tloušťka stěny profilu je moc velká. Dříve pro to existoval parametr, který tuto šířku definoval.
 * Chybějící podpora pro tisk místa pro adaptér. Bylo by vhodné namodelovat vlastní podporu než to nechat na slieru nebo celý airfoir
 * Díra na servo je moc nízká
 * Příčka pro připevnění serva je moc blízko vnitřnímu kraji
 * Pozice otvoru je špatná - páka serva vychází na spodní část směrovky

roman-dvorak, 2019-09-02:
Pokud jsou předchozí body nesrozumitelné, tak tady je rozepisuji:

1. to je asi jasné

2. Servo se do otvoru nevejde na výšku. Chybí tam asi 1 mm. Ale i pokud se to rozšíří o 1 mm na některou stranu, tak díry pro připevnění serva jsou ve špatné výšce. 
Na obrázku je nakreslen rozměr, kterým myslím výšku otvoru
[image]

3. Uvnitř otvoru pro servo jsou 2 "stěny" za které má být servo přišroubované. Připadá mi, že jsou úplně uprostřed. Je potřeba je posunout trošku k vnější straně směrovky. Důvod, tohoto posunu je, že nyní servo bude velmi vyčuhovat ven z profilu
[image]

4. Pokud bude servo v součastném otvoru směrem dolu, bude to vycházet tak, že táhlo bude na spodním okraji pohyblivé části směrovky. Bude to vycházet tak, že kabel od serva bude vždy na vzdálenější straně od tyče. Ale to se asi musí vyřešit šikovně udělanou cestičkou.

[image]

roman-dvorak, 2019-09-07:
Po vytištění směrovek se ukázalo několik problémů. 

 - [ ] Výplň v horní části
 - [ ] Silnější horizontální stěny v okolí podpor
 - [ ] Zaslepit otvor pro osu pohyblivé části

1) V části, kde je zešikmení pro sešroubování dvou polovin je problém, že neni namodelovaný infill. Infill je potřeba mít namodelovaný až do konce. Pokud se to tiskne bez infillu tak tam díky velkému zšikmení nedrží vrstvy po hromadě a je to velmi měkké.

2) V místech, kde jsou domodelované podpory (tedy v místě pohyblivé části a v místě, kam patří adaptér na tyč) jsou velmi tenké horní a spodní stěny. V těchto místech bych to navrhoval zesílit (alespoň na 2 - 3 vrstvy). Protože situace je taková, přemostění není moc kvalitní - jsou v něm velké mezery. A snadno se trhá. To i při vyřezávání podpor. 
Další vhodná úprava by bylo nad  a pod podporami udělat droubou mezeru, aby zde plast nebyl tak doře spojen.

3) Ve spodní části směrovky nemít díru pro osu směrovky až veskrz. Nechat pod ní pár vrstev (více než 1) Díky tomu se nebude muset zalepovat tavným lepidlem. Nahoře tyčky budo zajištěny tím, jak se směrovky sešroubují.

roman-dvorak, 2019-09-07:
Směrovka pro nový model je zřejmě dost malá. Je potřeba zvětšit její plochu. Vzhledem k tomu, že je využit celý tiskový prostor, musí se model rozdělit na více částí. Zároveň by se tím dal vyřesit problém potřeby modelovat podpory.

Představoval, bych si to tak, že se to rozdělí na 3 části. Spodní část, prostřední část - část, kde je pohyblivá část směrovky a třetí část by byl horní zbytek. Tyto 3 části by se následně slepily. popř. sešroubovaly.

Díky tomu je možné zvětšit výšku střední části do tiskárny a zároveň by bylo možné zvětšit hloubku profilu.

roman-dvorak, 2019-10-06:
Dneska jsme složili novou verzi směrovky. Ukázalo se několik problémů.

- [ ]  Nelze složit pohyblivá část směrovky s pevnou částí. Problém je, že díry pro tyčku jsou moc blízko konci výřezu pro pohyblivou část.
[image]

- [ ] Servo přesunout tak, aby kabel vedl rovnou do tyče. Nový prodloužený profil to dovoluje. Přinese to výrazně rychlejší skládání. 
[image]

- [ ] Použít jiný způsob modelování tohoto křídla. Jak to funguje, jsem se snažil popsat na [wiki](https://github.com/ThunderFly-aerospace/TF-G1/wiki/Modelov%C3%A1n%C3%AD-k%C5%99%C3%ADdel) Zřejmě to stále není dostačující. Takže jestli to z toho nechápeš, prosím napiš, pokusím se Ti doupřesnit, jak je to myšleno. 

- [ ] Všechno má strašně silné stěny. Stačil by jeden perimetr - pokud vyřešíš předchozí bod, tak toto nebudeš muset řešit. Protože to pak vyřeší slicer.

roman-dvorak, 2019-10-13:
- [ ]  Infill je dost hustý. Navrhuji zkusit vzdálenost výztuh nastavit na nějkých cca 130 % z aktuální hodnoty

- [ ] Odebrat díry pro vyztužení pomocí uhlíkových tyček. Zároveň tam pak nemusí bý ty plné výplně, které mají tyčky držet. Konstrukce se tím znovu odlehčí.

- [ ]  Pozice otvoru na servo je OK. Servo umístit blíže k vnější straně. Pokud servo vložím ze strany, kde je přepážka blíže ven, tak by to chtělo servo zapustit o 1 mm hlouběji.  Příčku, ketrá podepírá serrvo udělat tlustší (až na druhou stranu k vnější stěně profilu. Pro připevnění serva by se použily samořezné šrouby. Vzdálenost otvorů pro šrouby by měla být 28 mm. Průměr této díry ještě upřesním. 

- [ ] Na vnější straně směrovky zrušit díru pro servo. Servo se tam vloží i přišroubuje z vnitřní strany. Tato strana by měla být hladká a voděo odolná.

- [ ] Pro minimalizaci možnosti vniknutí vody do serva by servo mělo být natočeno směrem dolů.

- [ ] Na spodním dílu směrovky je díra pro osu směrovky vyvedena až ven. Je potřeba udělat její zaslepení. Zrychlí to montáž směrovky.a zjednoduší výrobní postup.

- [ ] Páka má stejný přesah i rozteč montážních děr jako její menší verze na staré směrovce. Takže tuto pozici táhla je nejlepší vzít odsud. Podle měření je tento přesat 5.5 mm od první montážní díry. 

[image]

- [ ] Na jednom středním dílu směrovky chybí otvory pro připevnění držáku na tyč. 

- [ ]  Odstranit tuto dílu na obrázku pod servem. Zřejmě nějaký pozůstatek po tunelu..

kaklik, 2019-10-22:
[image]

- [ ] šrouby adaptéru nesedí proti dírám ve směrovce
- [ ] servo je ve výklenku ze stran a zespodu hodně na těsno a je ho tam potřeba násilně vtlačit
- [ ] adaptér nepřechází plynule na plochu směrovky ale vzniká tam zub

roman-dvorak, 2019-10-28:
V servu je potřeba udělat úpravu, aby bylo možné osu směrovky snadno vytáhnout. Toto lze vyřešit odstraněním (snížením) kostky v okolí osy

closed 2019-11-03 by roman-dvorak

roman-dvorak, 2019-11-07:
@taury55 snažím se dostat směrovku do celkové sestavy vírníku. Jak nebyly dobře pochopeny připravené parametry, tak nefungoval původní výpočet vzdálenosti tyčí, na kterých je směrovka přidělaná. 

Teď jsem se snažil tento výpočet předělat tak, aby to sedělo na novou směrovku. Aby mi to sedělo, tak  k výpočtu musím přičítat nějakou divnou konstantu, aby to sedělo správně. Vidět je to např zde:

https://github.com/ThunderFly-aerospace/TF-G1/blob/improved_suspension/parameters.scad#L367

Nešlo by to nějak vyřešit?

taury55, 2019-11-11:
@roman-dvorak Dostal jsem se k tomu až teď, ale ta proměnná tail_pipe_distance je tam naprosto přebytečná a v celém 666_1028.scad není použita.

commit 4c81343c (2019-09-27): Rudder temporary height fix #194
commit 3e6b08c4 (2019-09-27): Rudder wire guide fix #194
commit d5694ebd (2019-09-27): Rudder bottom ribs fix #194
commit 47eeed6d (2019-09-27): Rudder moving part size fix #194