ThunderFly-aerospace/TF-G1 #195: Úchyt zadního kolečka na nápravnici

ChroustJan, 2019-06-16:
Pro testy na střeše potřebujeme udělat nový úchytný díl pro zadní kolečka. Díl by mohl vypadat následovně:
[image]

Díl přijde přidělat na zadní nápravnici o průměru 25mm. Stažení na tuto trubku by mělo být provedeno pomocí čtveřice šroubů M3. Do dílu následně přijde trubka o průměru 10mm na jímž druhém konci bude kolečko.

Není nutné tvořit díl označený číslem 2. Je možné mít proti sobě dva dílce 3.

alcamrkvicka, 2019-06-23:
888_2010
4 otvory jsou udelane na srouby M3x20, mezera mezi 2 castmi je 3 mm, je to tak ok?
[image]
[image]

roman-dvorak, 2019-06-23:
Tento díl mi příjde až moc zbytečně masivní. Navrhuji ho proto udělat tenčí
(a zároveň lehčí). Upevňovací šrouby dát max 1 cm od hlavní tyče. Na
straně, kam se vkládá tenčí trubka, tak udělat nějaký výčnělek nebo tento
díl udělat trochu oválný.

Myslím, že tento problém vychází z původního nákresu, který není nakreslen
úplně v měřítku.

[image: 20190623_183742.jpg]
(tento nákres také není 100 % v měřítku ale lépe popisuje poměr hlavní tyče
a sílu tištěných dílů.

tyto dvě části můžou být namodelovány v jednom souboru jako 2 moduly.

alcamrkvicka, 2019-06-23:
Ztenceno a upraveno do protahlejsiho tvaru...
[image]

ChroustJan, 2019-06-24:
Já to kreslil hodně zjednodušeně, aby to bylo jednoduché. Tento díl se mi samozřejmě líbí víc.

kaklik, 2019-06-24:
V modelu jsou [špatně užity parametry průměrů trubek](https://github.com/ThunderFly-aerospace/TF-G1/blob/81cb60f85d13ef3aa27e16925e3a3aaa7429a690/src/888_2010.scad#L6-L7). @ChroustJan  sice v zadání napsal průměry 10 a 25mm. To ovšem pro model není pravda a je potřeba použít [tyto parametry ](https://github.com/ThunderFly-aerospace/TF-G1/blob/81cb60f85d13ef3aa27e16925e3a3aaa7429a690/Parameters.scad#L3-L8)

roman-dvorak, 2019-07-01:
Při přimontování dílu byl zjištěn problém s tolerancemi ve velikosti parametru na 10 mm trubku. Trubka byla v tomto dílu volná. 

Díl se zdá pro tento účel použitelný.

ChroustJan, 2019-07-02:
Je potřeba posunout šroub držící trubku o síle 10mm do středu dílu. Viz obrázek. Takto dochází k vrtání do trubky na jejím okraji.
[image]

alcamrkvicka, 2019-07-02:
Dira je presunuta doprostred.

kaklik, 2019-07-02:
@ChroustJan  issue se uzavírá až když se otestuje jeho vyřešení. Do té doby je Resolved.

kaklik, 2019-07-05:
Nový úchyt funguje.

closed 2019-07-05 by kaklik