ThunderFly-aerospace/TF-G1 #196: Kryt vertikální trubky

Elizaa62, 2019-06-19:
- kryt musí plynule a vodotěsně přecházet na rotorovou hlavu

kaklik, 2019-06-21:
Co ten kryt ukončil tím zaobleným tvarem kapky, stejně jako třeba končí rotorový list?

Do tohoto tvaru by se teprve vyřízl otvor pro rotorovou hlavu. 

[image]

Elizaa62, 2019-06-27:
Zkusila jsem vložit tu kapku nahoru 
[image]

akorát je problém v tom, že se to nevejde do tisknutelného rozměru...

kaklik, 2019-06-27:
Tak to asi zbývá jen rozdělit na více dílů.

kaklik, 2019-07-30:
Toto řešení je pro současnou verzi ve větvi master dostatečné.

closed 2019-07-30 by kaklik