ThunderFly-aerospace/TF-G1 #197: Některé parametry z podložky 666_1027 přesunout do globálních parametrů

roman-dvorak, 2019-06-22:
V modelu podložky 666_1027 se vyskytují nějaké parametry (například ```z_size```) od kterého se odvíjí vzdálenost děr pro dvojici 10 mm tyčí. Tyto tyče se mají využít v novém návrhu ocasních ploch #194 

Kvůli tomu, že se tato hodnota použije i v jiném díle (666_1028 #194), tak je potřeba parametr přesunout do globálních parametrů.  

Tento název je trošku zavádějící, protože (mimo jiné) řeší vzdálenost (v ose Y - vůči letadlu) zmíněných děr a nemá to ve výsledném souřadnicovém systému nic společného s polohou v ose Z.

Elizaa62, 2019-06-27:
Parametr "z_size" se v 666_1027 vyskytuje jako parametr "z_size", ale v hlavním kódu pro podložku je jako parametr hull_z_size a tento parametr globálním parametrem je. Parametr je pro osu Z, ale ve výsledném souřadnicovém systému, jak píšeš, řeší osu Y.  Tato chyba bude na mé straně, protože jsem modely tvořila v nějakém souřadném systému a letadlo se skládalo v jiném.  Ráda bych parametr změnila tak, aby to souhlasilo s výsledným souřadným systémem, ale problém je v tom, že tento parametr se vyskytuje ve více modelech např. podložka, kryt horní i spodní a já netuším, co ta změna udělá...

kaklik, 2019-06-27:
Na tuhle změnu jde použít [nástroj v sublimetextu](http://docs.sublimetext.info/en/latest/search_and_replace/search_and_replace_files.html), který ti s tím pomůže. 

Možná je ale na tuhle operaci ještě lepší editor [Atom](https://atom.io/), který to podle mě umožňuje udělat [ještě jednodušším způsobem](https://flight-manual.atom.io/using-atom/sections/find-and-replace/). Je taky dobrý, že nemá žádnou otravnou hlášku s požadavkem na registraci.

kaklik, 2019-07-21:
Podložku je potřeba výrazněji upravit. Tyto změny zle tak provést při tom.

closed 2019-07-21 by kaklik