ThunderFly-aerospace/TF-G1 #198: Díl 888_1101 - úklid

Elizaa62, 2019-06-24:
Prosím autora o upravení kódu v dílu 888_1101!

Omlouvám se, ale nevím, co mají znázorňovat moduly "efsm_12 apod." Dále nevím z čeho přesně se skládá stl soubor dílu 888_1101.scad - prosím v kódu vyznačit. 

Dále se v kódu používají čísla, jejichž původ je neznámý, takže též prosím o vysvětlení.

Děkuji :)

roman-dvorak, 2019-06-24:
Na tomto dílu právě pracuji. Až bude dokončen, vše toto spravím. Je to nová varianta rotorové hlavy #169
 
Jinak EFSM-12: https://www.igus.cz/info/spherical-bearings-efsm

kaklik, 2019-06-24:
@roman-dvorak  Možná jen vysvětlím k čemu je to potřeba.
Aby mohlo být uzavřeno issue #193, tak je potřeba, aby rotorová hlava poskytovala některé náležitosti. 
Například musí být jasné, kde má být umístěn motor prerotátoru (jeho imitace) a jak mají být nakloněny listy s řemenicí. 
Cílem je, aby se nejasné číselné konstanty nešířily do dalších modelů.

roman-dvorak, 2019-06-26:
@Elizaa62  Já tady ale netuším, kde se bere tento posuv (těch 17.5 mm). Má to nějaký vetší význam nebo je to pouze posunutí od osy toho dílu? I rozteče těch děr pro přišroubování na trubku by chtělo definovat nějakým parametrem.

https://github.com/ThunderFly-aerospace/TF-G1/blob/1b70f4693953745a9728b779062dc351a7d923a0/src/888_1015.scad#L44-L57

Sestavu [888_1101](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/src/888_1101.scad) jsem přeuspořádal tak, aby to odpovidalo umisteni oproti konci svisle trubky. Ale je tam stále těch 17.5 mm nastaveno na pevno. V budoucnu by to mohlo dělat obdobný problém jako tyto mé konstanty.

Umisteni motoru ješte vytvořím jako funkci. Také to zkusím napočítat oproti konci trubky.

Jinak navrhuji vytvořit parametry, které budou definovat rotaci rotoru (a všech jeho častí), protože to může vypadat zajímavěji na různých datech pro grafika.

Vzhledem k tomu, že se vše rotuje kolem kulového čepu, tak navrhuji, že bude existovat funkce, která vrátí polohu čepu oproti konci trubky. A následně všechny další díly budou vztažené a rotované k tomuto bodu.

roman-dvorak, 2019-06-26:
[image]

Elizaa62, 2019-06-27:
Rozměr 17.5 značí posuv děr od osy. Je to čtvrtina rozměru 70 (nyní parametr "rotor_head_width" = 70). V kódu jsem to opravila na width/4.

Elizaa62, 2019-06-27:
V dílu 888_1101 chybí naklonění.

Elizaa62, 2019-06-27:
Je super, že je v dílu uveden model prerotátoru, ale do sestavy rotorové hlavy 666_2300 potřebuji vložit model pastorku řemenice.. jenže díl 888_1020  je od osy posunut na nějakou pozici a já nemůžu najít v kódu hodnotu o kterou je díl posunutý... Hodnotu potřebuji ke správnému umístění pastorku řemenice.

roman-dvorak, 2019-06-27:
@Elizaa62 upravil jsem tu Tvou sestavu. Zkus se na to podívat.

roman-dvorak, 2019-07-01:
@Elizaa62  Myslíš, že je toto issue dokončené? Nebo se po mne vyžaduje ještě nějaká úprava?

kaklik, 2019-08-06:
@Elizaa62  ping :)

kaklik, 2019-08-07:
@Elizaa62 Říká, že je to vyřešeno.

closed 2019-08-07 by kaklik