ThunderFly-aerospace/TF-G1 #199: Doplnit sestavu kolečka o vidlici 888_1027

kaklik, 2019-06-25:
Sestavu kolečka by bylo potřeba doplnit o vidlici 888_1027. Rozměry vidlice je potřeba nastavit tak, aby ke kolečku pasovala. Parametry vidlice by proto měly vycházet z rozměrů kolečka v souboru parameters.scad
 
[image]

alcamrkvicka, 2019-06-25:
[image]
 jak velka mezera ma mezi vidlici a koleckem byt?

kaklik, 2019-06-25:
Jestli myslíš z boku kolečka, tak žádná. Tam stačí na tloušťku té obruby vložek. 
Také jsem si všiml, že ty vložky v sestavě nejsou možná úplně správně zapuštěny v kolečku po úpravách jejich délky. 

Mezera mezi běhounem kolečka a zaoblenou částí vidlice stačí malá třeba 10mm. Ale do tohoto směru může být vidlice klidně delší, aby ji bylo možné uchytit za delší část trubky. Horním limitem je pak určitě tiskový rozměr 150mm.

kaklik, 2019-06-27:
Dalším požadavkem na vidlici je, aby využívala šrouby o nějaké existující délce. Pokud je to možné, tak například M4x50 nebo 55mm.

roman-dvorak, 2019-07-05:
Je potřeba upravit sestavu kolečka tak, aby poloha jednotlivých dílů byla nastavena pomocí odpovídajících parametrů a zkontrolovat jejich správnost. 

Do vidlice je potřeba použít šroub M4x55. Teď to vychází na délku šroubu M4x60 (který se moc nevyrábí). To lze vyřešit tak, že se z vložky ložiska odstraní vnější lem (viz: https://github.com/ThunderFly-aerospace/TF-G1/issues/191#issuecomment-508866092 ) a o dvojnásobek jeho tloušťky se zúží vidlice. Pak je potřeba ohlídat správné zapuštění matic (a hlavy šroubu) do vidlice na délku šroubu M4x55.

roman-dvorak, 2019-07-06:
Druhá část vidlice - třmen pro šroub není namodelován s použitím stejných parametrů.

roman-dvorak, 2019-07-07:
Zaktualizovány rozměry vidlice, upraveny připevňovací body, více zapuštěny matice pro prošroubování tyče, Délka šroubu napočínána pro M4x55 DIN 912.

closed 2019-07-09 by kaklik