ThunderFly-aerospace/TF-G1 #204: Zadní podvozek - úchyt podvozkových vzpěr k podélné tyči vírníku

roman-dvorak, 2019-07-01:
Tento díl slouží k připevnění pevných táhel k podélné tyči vírníku. Tento ze dvoučástí, které budou sešroubované skrze podélnou trubku těla vírníku. V podložce se vytvoří nové otvory, ve kterých se tento díl bude nacházet. 

Na něm bude přidělaná dvojice kulových ložisek [KSTM](https://www.igus.cz/info/spherical-bearings-pillow-block-kst). Jejich model bude vytvořen v rámci #201.

U ložisek je důležité, že jejich výchozí poloha musí směřovat směrem ke kolečku. Ložisko tedy musí být natočeno ve dvou osách. Toto natočení se musí určit ze základných parametrů nového podvozku. Dále se musí ohlídat dostatečná vzdálenost mezi těmito ložisky. To lze ověřit např ze sestavy. Tento dýl by měl být namodelován jednou. Při dalších použitích bude pouze otočen popř. zrcadlen.

při návrhu lze vycházet z dílů 666_1014 nebo 666_1017.

[image]

[image]

roman-dvorak, 2019-07-02:
[image]

roman-dvorak, 2019-07-03:
Vzhledem významným změnám v podložce #209 je možné upravit i připevnění tohoto dílu. V podložce by měly vzniknout dvě čtveřice děr (pro M4 šroub) ke kterým se sešroubuje placka s kulovými ložisky. 

[image]
 
[image]

ložisko musí být na placce rotováno ve dvou osách. Což není na tomto náklesu vidět.

Připevňovací díry
[image]

roman-dvorak, 2019-07-04:
Tento díl a díl #211 se zdá být tímto vyřešen. 
Nyní je potřeba v #209 vytvořit díry pro připevnění dílu. Je potřeba umístit vyztužovací tyče tak, aby vycházely přesně na střed tohoto dílu.

[image]

kaklik, 2019-07-30:
Aktuální řešení je zřejmě dostačující.

roman-dvorak, 2019-10-06:
Díl je potřeba upravit tak, aby obsahoval 2 díry do podložky. Aby se tak zamezila možnost otáčení dílu. Díra pro sešroubování musí být M3 (ne M4 jako doposud)

roman-dvorak, 2019-12-10:
Držák na tyče (ramena podvozku) je potřeba upravit tak, aby úhel děr pro sešroubování ramen měl stejný úhel jako díry v dílu #207. Trubka se pak bude snáze vrtat a bude stačit použít jednoduchý vrtací přípravek.

commit 43cd402a (2019-07-04): Kompletní přepracování dílu, zmenseni rozmeru, render

Resolved: #204