ThunderFly-aerospace/TF-G1 #205: Zadní podvozek - výběr vhodného tlumiče

roman-dvorak, 2019-07-01:
Jako vhodný tlumící člen podvozku #188 se jeví použití pneumatického pístu, který bude mít spojené obě části pomocí jednosměrného ventilu.

K vybranému pístu je následně nutné upravit připevňovací díl do Igubal ložisek. Stejně, jako je tomu u šestiosé platformy.

roman-dvorak, 2019-07-04:
Zajímavé řešení by bylo použití tlumiče s MRF (Magneto-Rheological Fluid) nebo ERF (Electro-Rheological Fluid) náplní. Výhoda tohoto řešení je, že lze elektronicky ovládat velikost tlumení.

Zatím jsem nalezl jediného výrobce MRF kapaliny [Lordmr](http://www.lordmrstore.com/lord-mr-products/mrf-132dg-magneto-rheological-fluid-250ml). Také nabízí rovnou tlumiče s touto kapalinou a integrovaným magnetem [RD-8041-1](http://www.lordmrstore.com/lord-mr-products/rd-8041-1-mr-damper-long-stroke). Díky tomu, že je magnet umístěný uvnitř pístu, tak by nemuselo docházet k velkému úniku magnetického pole mimo obal. Tím následné rušení magnetometru.

kaklik, 2019-08-19:
Zatím se zdá, že pneumatické tlumení je dostatečné.

closed 2019-08-19 by kaklik