ThunderFly-aerospace/TF-G1 #206: Zadní podvozek - připevnění tlumiče k příčné podvozkové tyči

roman-dvorak, 2019-07-01:
Na příčnou tyč je potřeba kyvně přichytit pneumatický píst. K tomu je vhodné znovu použít kulová ložiska Igus. Naklonění ložiska bude vypočítáno z celkových parametrů podvozku.

[image]

[image]

Lze vycházet z dílu #195

roman-dvorak, 2019-07-20:
Vybrat vhodnou délku šroubu pro připevnění kulového ložiska k dílu a do dílu zapustit matku místo hlavičky. Nebude tam pak ani muset být servisní otvor pro imbus klíč. 

Díl je zbytečně robusný, je potřeba sjednotit délku šroubu pro stažení tyče.

roman-dvorak, 2019-07-22:
Díl bude předělán na boční uchycení na trubku.

roman-dvorak, 2019-07-22:
[image]

kaklik, 2019-08-11:
K dílu pouze chybí vrtací přípravek, jinak je díl použitelný.

closed 2019-08-11 by kaklik

roman-dvorak, 2019-08-13:
Vrtací přípravek již nějakou dobu existuje v této sestavě: https://github.com/ThunderFly-aerospace/TF-G1/blob/improved_suspension/STL/print_plate_888_2016.stl

Sestavu jsem ještě přerenderoval

commit d166268e (2019-07-22): Resolved: #206 optimalizace hmotnosti + render