ThunderFly-aerospace/TF-G1 #207: Nový podvozek - připevňení vzpěr a tlumiče ke kolečku

roman-dvorak, 2019-07-01:
Mezi vidlicí a tyčemi bude mezikus, který bude sloužit ke spojení obou tyčí od podvozku a tlumiče. Dvě tyče budou k tomuto dílu připevňeni na pevno tak, aby nedocházelo k jeho naklánění a nějakým deformacem. Pravděpodobně by tyče měly být přidělány prošroubováním šroubu skrze ně k této destičce. Je potřeba, aby držáky tyčí měly správný úhel. 

Dále k tomuto dílu bude přidělán kulový čep z řady Igubal, za který bude přidělán pneumatický tlumič. Domek tohoto ložiska bude muset být přidělán také trochu šikmo. 

Tvar celé "destičky" je zatím volný. Nemusí vůbec odpovídat nákresu. Je to vhodné dělat podle potřeb vidlice s integrovaným motorem/brzdou. To samé platí o připevňovacích dílech. 

Rozdělení těcho dvou dílů je z důvodu snazšího tisku a následného vylepšování vidlice. 

[image]

[image]

roman-dvorak, 2019-07-20:
Kulové ložisko KBRM má větší vnější průměr (než EFSM) a nevejde se tedy do připravené díry. Dále by bylo vhodné tento díl udělat kompaktnější tím, že nebude obsahovat jednotlivé trubyčky ale bude to obalené obálkou.

roman-dvorak, 2019-07-21:
Nalzené problémy opraveny. 

[image]

kaklik, 2019-08-11:
Tento díl myslím funguje. :)

closed 2019-08-11 by kaklik