ThunderFly-aerospace/TF-G1 #208: Nový podvozek - upravená vidlice s integrovaným motorem

roman-dvorak, 2019-07-01:
Protože zadní podvozek bude sloužit k zatáčení vírníku, je potřeba do podvozkové vidlice integrovat motor, který bude vždy jedno kolečko přibržďovat.

Mělo by to být řešeno tak, že se plášť kolečka bude dotýkat s rotačním obalem motoru. 

Vidlice bude následně připevněna na díl #207

roman-dvorak, 2019-07-21:
Vidlice musí mít obráceně umístěný šroub tvořící osu kola. Když se vidlice přišroubuje k dílu #207 tak nelze osu kolečka vytočit/připevnit. Matka může být umístěna směrem k #207. Tím dojde k tomu, že matka nevypadne.

kaklik, 2019-08-20:
Zde by zřejmě bylo ještě vhodné: 

- [ ] posunout náboj kola mezi dva montážní šrouby dílu #207 - Tím dojde ke zmenšení namáhání dílů, neboť síly budou při roztahování podvozku působit blíže osy kola. Navíc nebude vypadávat matice šroubu tvořícího osu kola.
- [ ] Doplnit kryt na enkodér, kabely a přečnívající osu motoru.

roman-dvorak, 2019-08-27:
- [ ]  Přes enkodér namodelovat nacvakávací kryt, který ho bude chránit před prachem atd..
- [ ]  Využít všechny 4 díry motoru
- [ ]  Zkrátit vidlici - je zbytečně vysoká
- [ ]  Napočítat délku šroubu do páky s motorem

roman-dvorak, 2019-09-03:
Motor nad vidlicí může být ke kolečku dotahován pomocí zkrutné pružiny, která bude obmotána kolem osy páky pro motor. Pružina bude schovaná mezi dílem vidlice a držákem motoru.

Dojde tím zároveň k hmotnostní optimalizaci držáku

roman-dvorak, 2019-09-16:
Jako vhodná síla pro přítlak motoru ke kolu mi připadá něco mezi 15 - 20 N. Páka je na 5cm. To znamená 0.75 - 1 Nm. 

Chod pružity k vytvoření této síly by měl být cca 20-30 stupňů.

Měření této síly bylo provedeno tak, že jsem kolečko vírníku postavil na váhu a rukou jsem přitlačoval motor oproti kolečku směrem dolů. Tato síla se pak změřila na váze.

roman-dvorak, 2019-10-06:
Otvory pro páku s motorem byly prozatím odebrány. Vidlice byla snížena a mírně upravena poloha šroubů pro přípevnění k podvozku

closed 2019-10-06 by roman-dvorak