ThunderFly-aerospace/TF-G1 #209: Úpravy podložky 1027

roman-dvorak, 2019-07-02:
Pro nový podvozek #188 #189 a novou směrovku #194 a nový způsob přichycený rotorové hlavy je potřeba provést několik úprav podložky. 

- [ ] Implementovat připevňení podvozkových ramen skrze KSTM ložiska
- [ ] Zadní příčnou tyč přidělat přímo do podložky
- [ ] Odebrat střední hliníkovou tyč
- [ ] Jako kostru podložky použít 2 10 mm tyče
- [ ] Vytvořit otvory pro připevňení směrovek a jejich prokabelování
- [ ] Vytvořit připevňovací otvory pro přední kolečko, kotvu magetu, payload a další (lehké příslušenství.

kaklik, 2019-08-06:
Podle mě se tohle musí rozdělit na jednotlivé issue vycházející z #223

roman-dvorak, 2019-09-01:
Toto issue vyřešeno pomocí #223

closed 2019-09-01 by roman-dvorak