ThunderFly-aerospace/TF-G1 #221: Nový pylon

roman-dvorak, 2019-07-30:
Uchycení rotorové hlavy bude nyní provedeno pomocí čtveřice uhlíkových tyčí uspořádaných ve tvaru hran čtyřbokého jehlanu. Cílem tohoto úkolu je odlehčení vírníku a zároveň zvýšení pevnosti.

Tyče budou zastrčené v podložce, ke které budou nějakým způsobem připevněné a zajišťěné proti pohybu. 

Rotorová hlava bude složena ze 3 částí.

Spodní část je držák serv. Bude to jednoduchý díl, který nebude mít (zatím) žádné jiné využití. Možná by se k němu mohla přidělat venturiho trubice, pitotka, atd.. 

Druhý díl bude spojovat všechny 4 tyče. Bude to taková kostka, do které se z boku zastrčí ty jednotlivé tyče a přes vytištěnou podložku se to sešroubuje dohromady.

slimonslimon, 2019-08-19:
> Tyče budou zastrčené v podložce, ke které budou nějakým způsobem připevněné a zajišťěné proti pohybu.

jak se dá přidělat uhlíková tyč? dá se provrtat pro pojišťovací šroubek? nebo jen na stisk?

> ude to taková kostka, do které se z boku zastrčí ty jednotlivé tyče a přes vytištěnou podložku se to sešroubuje dohromady.

Tohle zatím úplně nechápu

roman-dvorak, 2019-08-19:
Přichycení té tyče už mám vyřešené. Bude to částečně zasunuté a částečně sešroubované skrz trubku. 

Můj díl končí plackou nahoře. 
[image]

Nad tím by měl být Tvůj díl rotorové hlavy (včetně ložiska) který se k tomuto pouze přišroubuje. To se  musíme dohodnout na rozteči děr atd.. Ale to je velmi variabilní. Takže bych to viděl podle toho, jak Ti to vychází.

Výchozí záklon (cca 9deg) by měl být řešen ve Tvém dílu. 

Serva rotorové hlavy budou umístěna pod tímto spojovacím dílem. Je potřeba to napočítat tak, aby délka táhel byla 50 nebo 75 mm.

kaklik, 2019-08-19:
Myslím, že ten díl ukončující uhlíkové trubky by měl objímat trubku po celém obvodu, aby tam trubka kromě prošroubování mohla být i zalepená.

roman-dvorak, 2019-09-16:
Tato konstrukce pylonu pro rotorovou hlavu se zdá dostatečně robutní. Celá konstrukce se zjednodušila a 2 menší díly. 

https://github.com/ThunderFly-aerospace/TF-G1/blob/6e5d2e93115e60de0cdc139af84ed9f61570eb04/STL/888_1029.stl

closed 2019-09-16 by roman-dvorak