ThunderFly-aerospace/TF-G1 #223: Nová konstrukce nosné podložky

ChroustJan, 2019-08-04:
Při zvětšování nosné podložky dochází k problému nutnosti dělení dílce na tisknutelné části. Tento proces není příliš jednoduchý. Stávající podložka se také tiskla kolem 80h. Při vývoji a testech je potřeba co nejsnáze měnit rozložení vnitřní výbavy vírníku. Přibývá nová elektronika, nebo je potřeba posunout například baterie kvůli těžišti. 

Nová konstrukce má pouze dva hlavní dílce:
- tuhý středový trup skrz který vede dvojice uhlíkových trubek (v nákresech 1)
- okrajový dílec skrz který vede jedna uhlíková trubka, dílec také obsahuje drážku pro uchycení krytu vírníku jako tomu bylo u současného řešení ( v nákresech 2)

Nákres ze shora:
[image]

Konstrukce umožňuje zachovat stávající díl předního motoru. Pouze bude potřebovat upravit uchycení na dvojici trubek. Plánovaný ocas se také nemění. Pouze bude vytvořen dílec pro uchycení dvojice trubek z ocasu na dvojici trubek rámu vírníku. Konstrukce zakrytování vírníku bude také zachována. Pouze bude jiné rozložení otvorů pro úchytné šrouby.

[image]

Náčrtek dílce hlavního nosného trupu. V rámci iterací bylo rozhodnuto, že se ocas nebude chytat na nosné trubky, ale na pomocné boční. Viz. první obrázek pohledu na vírník z vrchu. 

[image]

Podrobnější rozkreslení dílce 1 a 2:

[image]

Dílec 1:

- jeho délka by měla být parametricky rostoucí. Maximální délka je omezená velikostí 

- tiskárny.

[image]

[image]

roman-dvorak, 2019-09-16:
- [ ] Sjednotit rozměry děr (a příslušných matek) skrze nosník
- [ ] Do sestavy přidat značky pro jednoduché určení dílů
- [ ] Odebrat podpory pro tisk univerzální drážky.

roman-dvorak, 2019-09-23:
Nosník se zdá použitelný. Další problémy budou mít vlastní issue.

closed 2019-09-23 by roman-dvorak