ThunderFly-aerospace/TF-G1 #224: Implementace podélného dělení horního krytu

kaklik, 2019-08-05:
Horní kryt je potřeba rozdělit i podélně na menší tiskové díly, aby bylo možné vytisknout trup širší, než je tiskový rozměr 3D tiskárny.

kaklik, 2019-08-06:
Dělení krytu funguje přesně tak jak má a lze vybrat jen část krytu. Například: 

[image]

Přestaly ovšem díky tomu fungovat existující tiskové soubory [print_plate_1025X.scad](https://github.com/ThunderFly-aerospace/TF-G1/tree/master/src/4printing). Ty je potřeba poupravit tak, aby bylo možné samostatně vytisknout každý díl krytu. 

Je potřeba při tom pamatovat na to, že maximální tiskový rozměr tiskárny takto dělený kryt nejlépe vyplní úhlopříčně.

kaklik, 2019-08-06:
Zkusil jsem si změnit parametry, abych zjistil, jak se dělení chová.  Přišel jsem na to, že nová podélná výztuha, se při aktuální kombinaci parametrů nechová úplně správně a vyčuhuje mimo trup. 

[image]

Elizaa62, 2019-08-07:
Nefungovalo to, protože jste změnili parametr délky kapky z "hull_drop_lenght" na "hull_x_size". Už by to mělo fungovat.

Elizaa62, 2019-08-07:
Jaké parametry tam měníš?

Elizaa62, 2019-08-07:
Už by to mělo fungovat :)

Elizaa62, 2019-08-07:
> Dělení krytu funguje přesně tak jak má a lze vybrat jen část krytu. Například:
>
> [image]
>
> Přestaly ovšem díky tomu fungovat existující tiskové soubory [...]

-> ve které větvi mám toto udělat? Master nebo improved_suspension?

kaklik, 2019-08-07:
> Jaké parametry tam měníš?

Tak hýbal jsem s hull_x_size a hull_drop_lenght protože jsem nevěděl jaký je mezi nimi rozdíl. 

Teď ale můžu potvrdit že to funguje. :) 

[image]

A z těch [dvou parametrů jsem udělal jeden](https://github.com/ThunderFly-aerospace/TF-G1/blob/0a92604a2d136537b65ac377492dbab5660f56b0/parameters.scad#L161), tak že jsem hull_x_size přiřadil hodnotu parametru hull_drop_lenght s tím, že parametr hull_drop_lenght bude potřeba v budoucnu zřejmě úplně zrušit.

kaklik, 2019-08-07:
> -> ve které větvi mám toto udělat? Master nebo improved_suspension?

Tohle oprav ve větvi improved_suspension.  V master stále funguje původní varianta krytů a tam ji není potřeba měnit.

Elizaa62, 2019-08-13:
> Tohle oprav ve větvi improved_suspension. V master stále funguje původní varianta krytů a tam ji není potřeba měnit.

--> jak myslíš úhlopříčně? Stačí otočit o 45 stupňů nebo ten díl mám ještě rovnat přesně na střed tisknutelného rozměru?

closed 2019-10-28 by roman-dvorak