ThunderFly-aerospace/TF-G1 #225: Tisknutelný kloubový spoj rotorové hlavy

kaklik, 2019-08-08:
Současná tisknutelná rotorová deska, viditelná na následujícím obrázku funguje relativně dobře.
Díl se nyní naklání volně obou osách (podélně-pitch i příčně-roll). Problém je dost vysoká volnost okolo vertikální - Yaw osy. 

[image]

Existují dva možné přístupy k tomuto problému. 

- Buď opustit hlavu v provedení spindle head, která má pro 3D tisk výhodnou skladbu a navíc umožňuje teoreticky let na zádech a nahradit tuto konstrukci klasickým uspořádáním [offset gimbal head](http://www.msgyro.com/photos/Bensen_Gimbal_Head_Hub_Bar_and_Head_Plates_504x379.jpg)
- Nebo upravit tvarový kloub bránící protočení hlavy.

slimonslimon, 2019-08-08:
mám v hlavě návrh kolem toho kulového kloubu udělat tisknutý kardanův kloub.

ten kulový kloub umístit na vyvýšenou plošinku, která by měla v jedné ose očka přesně ve výšce kloubu, pak druhý díl, který by to obepínal zvenku (mohl by mít shora plošku se zářezem pro tyčku, což by mu zvýšilo pevnost) a třetí díl (vlastní deska s prerotátorem) by měla dvě očka smerem dolů a obepínala by s offsetem ten prostřední díl - kromě oček by s tohoto dílu musel být vytištěn ještě krček kolem osyčky, který by zajišťoval stažení horního dílu proti kulovému kloubu

slimonslimon, 2019-08-08:
přes kardanův kloub by neměli jít v podstatě žádné síly, krom kompenzace yaw

slimonslimon, 2019-08-08:
[image]

kaklik, 2019-08-08:
Jo, tohle hybridní řešení se mi docela líbí. Pokud to navíc cené nevyjde moc masivní, tak v tom nevidím problém. 

Pokud to správně chápu, tak ty otočné části ani nemusí být spojeny osou ale stačila by jen břitová vazba,  Podobně jako to máme na [vyvažovacím přípravku](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/manufacturing_helpers/888_1009.stl).

slimonslimon, 2019-08-08:
břitová vazba zní zajimavě, bylo by to jednodušší na složení a asi i kreslení, akorát nevím jak dobře drží proti rotaci (yaw).

kaklik, 2019-08-09:
> břitová vazba zní zajimavě, bylo by to jednodušší na složení a asi i kreslení, akorát nevím jak dobře drží proti rotaci (yaw).

Tak bude to držet tak moc, dokud se ty břity neulomí. Ovšem vzhledem k tomu, že tady nejde úplně o nutnost malého odporu spoje, tak by břity ve skutečnosti mohly mít i poměrně nízký rozsah úhlů. Pak by pevnost vazby nebyla dána tolik pevností břitu, ale pevností kulového ložiska proti vytažení kuličky. (klín břitového spoje by při krutu v ose yaw působil silou roztahující celý kulový spoj.)
Zřejmě zde bude existovat nějaké optimum pevností. Neboť je výhodnější, aby se při havárii hlava protočila a nikoliv utrhl celý rotor.

kaklik, 2019-08-19:
Kloubový spoj funguje a nyní je potřeba jej implementovat do nové verze rotorové hlavy.

closed 2019-08-19 by kaklik