ThunderFly-aerospace/TF-G1 #228: Pročištění repozitáře

roman-dvorak, 2019-08-13:
Před zásadními úpravami konstrukce vírníku chci provést úklid repozitáře. Má to několik důvodů:

- Existuje mnoho model, které se na aktuální verzi vírníku nepoužívají
- Mnoho dílů je pojmenováno špatně. Je potřeba provést pojmenování dle této konvence [openscad - nazvy souboru](https://wiki.mlab.cz/doku.php?id=cs:openscad#nazvy_souboru)

Tyto úpravy budou provedeny ve větvi "improved_suspension". 

Po dokončení tohoto issue bude potřeba kompletně přerenderovat všechny soubory. Zároveň bude zaktualizován makefile od @JanKott :)

roman-dvorak, 2019-08-13:
Změny: 

- Odstraňen 888_1002

kaklik, 2019-10-21:
Bylo by také potřeba provést úklid v dílech rotorové hlavy. Což znamená 

- [ ]  smazat testovací vývojové díly
- [ ] zajistit, aby v master větvi byly díly rotorové hlavy pro vírník na střechu auta a v improved_suspension díly kusu pro letové testy. 

V master mají být zřejmě zdrojáky k[ této variantě rotorové desky](https://github.com/ThunderFly-aerospace/TF-G1/blob/b2b7fd36e55d263fe05340220915ee90a58ea3cf/STL/888_1020.stl) (M3 šrouby táhel, neskosené hrany u táhel)

slimonslimon, 2019-10-22:
snad se mi to povedlo. 
Horní díl rotorové hlavy je takřka totožný v obou větvích, liší velikostí šroubků serva - je to vytaženo do paraetrů na začátku souboru.