ThunderFly-aerospace/TF-G1 #229: Držáky kabeláže

ChroustJan, 2019-08-15:
Navrhnout univerzální držáky kabeláže pro novou koncepci vírníku, aby šli dát do drážek v trupu.

kaklik, 2019-08-15:
Ještě by to sem chtělo doplnit ty obrázky...

ChroustJan, 2019-08-15:
Přibližný profil trupu:
[image]

Náčrtky koncepce držáku:
[image]

[image]

roman-dvorak, 2019-08-27:
Tohle bude prozatím vyřešené použitím "Flexibilního kabelového držáku", který byl koupený původně do auta. Uřízne se jeden nebo dva háčky a do spodní strany se vyvrtá díra pro M3 a přišroubuje se do univerzálních podélných drážek.  Popřípadě se může zkusit využít samolepka na tomto držáku.

[image]

closed 2019-08-27 by roman-dvorak