ThunderFly-aerospace/TF-G1 #231: Vytvořit univerzální vrtací přípravky

roman-dvorak, 2019-08-18:
Udělat vrtací přípravky, které budou obsahovat díry pro provrtání skrz střed tyčí tak i pro boční "čepy" pro uhlíkové tyče. Přípravek by měl mít stahovací šroub, aby bylo možné ho pevně umístit na tyč a stáhnout pomocí šroubu.

Bylo by fajn otestovat přípravek, který bude obsahovat vložky z provrtaných matic.

alcamrkvicka, 2019-08-26:
Dil je nazvany drill_tool a nachazi se v manufacturing_helpers.
Na zacatku je mozne zakomentovanim zvolit prumer tyce (25mm nebo 10 mm) a sroub (M3 nebo M4).
[image]
Je to takto v poradku?

kaklik, 2019-08-28:
Ještě by bylo potřeba k dílu vytvořit tiskové soubory, které budou obsahovat díl natočený v tiskové pozici (otvorem pro trubku vertikálně) a budou obsahovat různé kombinace průměrů šroubu a trubky.

roman-dvorak, 2019-09-01:
Dále by bylo potřeba vytvořit varianty, které budou mít na okrají vrtacích děr šestihranný otvor pro zastrčení matic s provrtaným závitem. Cílem toho je, aby vrtách procházel nějakým tvrdším materiálem a neopotřebovával se pak tolik vrtací přípravek. 

Těsně u trubky by následně mohlo být znovu rozšíření, aby se špony nemačkaly do plastu.

closed 2019-11-21 by github-actions[bot]