ThunderFly-aerospace/TF-G1 #238: Varianta desky rotorové hlavy s kardanem

roman-dvorak, 2019-09-07:
V návaznosti na #149 #169 a #225 se ukazuje, že tisknutelná deska rotorové hlavy má několik prolémů, které by bylo možné vyřešit frézovanou deskou tak, jak tomu bylo u #169. 

Problémy byly především v tom, že je těžké vytisknout dostatečně tuhou desku. Především od krčku, kudy prochází osa rotoru po táhla od serv. Další problém je, že plastová deska neodvede od motoru prerotátoru dostatek tepla a pak se plast natavuje. Čímž se zmenšuje vzdálenost řemenic a tedy napnutí řemene. Ten následně přeskakuje.

Původní řešení mělo problém, že čep, který měl držet rotor v konstrantním Yaw úhlu se vymačkal do plastu a byla tam velmi velká vůle. Proto vznikalo několik variant řešení tohoto kloubu. Následně @slimonslimon přišel s řešením, které se zdá jako funkční. Použití obdoby kardanova kloubu. V YAW je zde vemi malá vůle. Vytištěné řešení má zmíněné problémy.

Je varianta, že deska by byla stále frézovaná z 5 mm duralové desky AW7075 a kardan by byl vytištěný a přišroubovaný k této desce. V desce by byly z horní strany dva M3 šrouby a v kardanu (v nejvyšší části) by byla kapsa na matky.  Tím by se docílilo dobré tuhosti dílu a zároveň využití kardanu pro účely omezení rozsahů pohybu a úhlů.

Deska by mohla vypadat nějak takto:
[image]

Ze spodní a z hodní strany by byly 2 soustružené díly, které by vymezovali správnou vzdálenost mezi deskou, kulovým ložiskem a ložisky rotorové hlavy.

slimonslimon, 2019-09-08:
nakrelsit plastový díl který by se na tu desku přišrouboval je triviální - defacto by to byl díl který vznik pro test kardanu (bez čísla v nějaké kardanové větvi). Akorát by měl uprsotřed dáru na ty soustružené díly. 

trochu otázka je, jak naložit s výškou toho krčku.

slimonslimon, 2019-09-08:
mimochodem, u současného řešení je také duralová deska - jak je vyřešeno usazení na to kulové ložisko?

Defacto by se pro potřebu testu dala současná deska upravit...

roman-dvorak, 2019-09-08:
Tam není žádný krček. Deska leží skoro přímo na kulovém ložisku. Je tam pouze vytištěná podložka/vložka ložiska, která má tloušťku něco okolo 0.5 mm. Nad deskou jsou jen okolo osy 2 podložky a hned ložiska. 

[image]

slimonslimon, 2019-09-09:
Ještě další otázka : V #169 je zmíněna varianta z duralového plechu - to je slepá vývojová větev?

roman-dvorak, 2019-09-09:
Na obrázku v mém minulém komentáři je právě ten plech, který má výkres v #169. Náhrada za tisknutelný díl se zdála v té době vhodná z důvodu, že to umíme snáze vyrobit. 

Následně pak přišlo několik variant tisknutelné hlavy a konečně ta tvá se zdála použitelná. Při namontování na vírník a následných testech se objevilo několik nových problémů.

Takže já jsem velmi silně pro tyto dvě řešení spojit a vzít z nich ty dobré části.. :) Jako základ frézovaný plech a jako kloub použít kardan.

roman-dvorak, 2019-09-16:
O víkendu jsem se stavil novou verzi rotorové hlavy. Konstrkce je zajímavá a docela dost pevná. Problém je v uchycený tyčí do podložky #241. 

[image]

Toto řešení má několik problémů:

Snížením profilu desky je deska podstatně měkčí. Díky tomu ji velmidokážu ohnout tak, že se kolo desky dotkne. To by měl vyřešit duralový plech. Vypadá to, že výška krčku půjde snížit na minimum. 

Další problém je, že nyní k tomu nelze připevit igus táhla. Důvodem je velká objímka kuličky. Objímka nyní koliduje s řemenicí či řemenem.

Pevnost v Yaw se zdá dostatečná. Ale to je pouze na základě testů v ruce.

slimonslimon, 2019-09-16:
vzhledem k tomu, že rámeček kardanu mám možnost se opřít při zkoucení o podpůrné okraje, tak to asi rozlámat jen tak nepůjde, ale nechme se překvapit.

nabízí se varianta, že by na duralovou desku byly přidělány tisknuté prvky na přidělání táhel - to by se dalo zkombinovat s tisknutými zuby a boky kardanu. 
Duralový plech by pak nemusel být frézovaný, mohl by být pouze vrtaný o konstantní šířce- zezdola by k němu byl přišroubován tištěný díl.

roman-dvorak, 2019-09-16:
Jojo, o tom jsem také přemýšlel, že by to táhlo bylo přidělané přes nějaký vytištěný díl. Že by tato tištěná "kostka" byla přišroubovaná stejným způsobem jako kardan a pak by obsahovala otvor pro připevňení táhla.

Je pravda, že pro první verzi by stačil vzít duralový plech a z něj to vyvrtat. Jako jediný problém vidím s přesným vyvrtáním té díry pro osu. Aby to mělo správné tolerance. Zatím by tam nebylo řešěno odlehčení dílu atd.. Nebo jen nějak dočasně.

roman-dvorak, 2019-10-07:
- [ ] Střední díl kardanu má okolo sebe velmi velké tolerance. Důsledek toho je, že kardan má značnou volnost v yaw ose. Tolerance můžou být poloviční.

- [ ] Rohy středního dílu kardanu udělat kulaté. To by mělo zvýšit odolnost rámečku proti prasknutí a vyšší pevnost v rozích.

- [ ] Do desky kardanu vytvořit otvory pro možnost utáhnout šrouby držicí ložisko bez nutnosti demontáže celé hlavy.

kaklik, 2019-10-14:
Nová konstrukce je funkční.

closed 2019-10-14 by kaklik