ThunderFly-aerospace/TF-G1 #245: Vytvořit zatáčecí mechanizmus na přední kolečko

roman-dvorak, 2019-09-23:
Protože zatáčení pomocí VESC regulátorů nejspíše zabere ještě nějaký čas, bylo by dobré si vytvořit možnost řízeně pojíždět i s jiným způsobem řízení.

Řízené by bylo přední kolečko. Na díl pod nosníkem je potřeba vymyslet způsob připevnění serva. První otáčející se díl musí mít na sobě nějakou páku nebo místo pro její připevnění.

kaklik, 2019-10-01:
@roman-dvorak  nechybí v té páce zapuštění pro matku šroubu táhla? 

Ta tloušťka stěny pod šřouby, které tu páku mají držet se mi nezdá zrovna velká.

roman-dvorak, 2019-10-06:
Spodní část to chce ještě posunout výše. Jinak je to funkční řešení.

roman-dvorak, 2019-10-08:
Vyřešeno otočením dílu. Změnou polohy zapuštění pro matku.

closed 2019-10-13 by roman-dvorak