ThunderFly-aerospace/TF-G1 #262: Zpevnění předního kolečka

roman-dvorak, 2019-10-29:
Při testu se ukázalo, že přední kolečko #189 není dostatečně tuhé a kolečko se z dílu dokázalo vylomit.
[image]

Při prohlížení dílu se zjistilo, že trubička okolo osy není s ničím zpevněna. Pouze infillem. Tento problém je potřeba eliminovat několika změnami. 

- [ ] Použití delšího lícovaného šroubu 8x70.
Tato změna by měla sílu rozložit na větší plochu. Ale i tak by v  dílu byla trubička, která nebude ničím pořádně vyztužena. To je potřeba vylepšit dvěmi způsoby.

- [ ] Prodloužení kapsy na matku.
Kapsu na matku lícovaného šroubu by bylo vhodné udělat skrz celý díl. V orientaci zepředu dozadu. 

- [ ] Vyztužovací otvory u osy kolečka. 
V dílu vymodelovat výztuhy od osy kolečka. Takové výztuhy mají za cíl v dílu vytvořit perimetry, které budou spojovat trubičku osy s vnějšími perimetry. To může být provedeno tak, že skze model budou vytvořeny díry (např d=1mm), které budou následně umístěny do spirály podél lícovaného šroubu. 

Mělo by to být uděláno tak, že následující díra bude pootočena např o 110 stupňů a o cca 3 mm výše.

- [ ] Mezi prvním a druhým dílem předního podvozku (tam, kde se podvozek otáčí) je potřeba do horního dílu zapustit podložku. To má za cíl zvětšit plochu, přes kterou se přenáší síla a díky zapuštění nebude mezi díly taková mezera. A při větší síle se styčné plochy těchto dílů o sebe opřou a nebudou tolik vylamovat osu.

Tyto body jsou znázorněny na následujícím nákresu:
[image]
Na prvním obrázku je vidět vnitřní struktura dílu. Na druhém je pak řez při pohledu z boku.

kaklik, 2019-10-30:
Alternativní možností, jak držák kolečka zpevnit je udělat ho žebrovaný. Podobně, jako byl [zpevněn krček rotorové desky](https://github.com/ThunderFly-aerospace/TF-G1/blob/f3a17a0dad45fde1ed40cace4f3c18f7caf67669/STL/888_1020_plastick_neck.stl).

kaklik, 2019-11-12:
Šrouby držící horní díl předního kolečka musí mít matky až pod kapsou pro matku hlavní osy kolečka. Cílem je aby v žádné části dílu nebyly síly přenášeny tahem za vrstvy ale, tlakem. 

Nejdelší M4 šrouby, které aktuálně máme jsou M4x55 mm dellší by bylo nutné koupit.

roman-dvorak, 2019-11-12:
No, aktuálně tam jsou použity šrouby M4x55.

kaklik, 2019-11-13:
Také je potřeba, aby model byl uzpůsoben na kratší páku serva. Nově bych tam dal páku 25mm.

closed 2019-11-19 by kaklik