ThunderFly-aerospace/TF-G1 #264: Kryt na akumulátory

roman-dvorak, 2019-11-02:
Na baterie je potřeba ze spodní strany vytvořit "krabičku", která bude akumulátory chránit před případným poškozením. Krabička by se mohla i nějak zacvaknout do držáku #250. Ale to není, v první iteraci potřebné.

Měla by to být jednoduchý kvádr, který se nasadí na akumulátor ze spodní strany a bude obsahovat otvory pro dráty.

kaklik, 2019-11-11:
Myslím si, že kryt na akumulátory by měl být spíše řešen jako spodní kryt, celého vírníku. Neboť akumulátory je potřeba temperovat a do spodního krytu by bylo možné vhánět vzduch.

roman-dvorak, 2019-12-09:
V návaznosti na #275 bude spodní kryt ve vlastní drážce. Je tedy potřeba upravit kryt, aby obsahoval tento "pásek". Vzdálenost od středu krytu (z=0) je v parametru ```bottom_cover_strip_zposition```. 

Nová funkce pro tvorbu děr jsem do krytu už implementoval. 

Bude zřemě potřeba, podle sestavy, pak také upravit díru na přední podvozek - resp. servo.