ThunderFly-aerospace/TF-G1 #265: Aktualizace celkové sestavy

kaklik, 2019-11-03:
Celkovou sestavu vírníku je potřeba aktualizovat podle nových dílů. 

- [ ]  doplnit o přední kolečko
- [ ] zarovnat kryty a díry v lemu #260
- [ ] Vyrenderovat sestavu

roman-dvorak, 2019-11-06:
@cesilko1  Mohl bych Tě poprosit o slozeni predniho kolecka? Vůbec se tam nevyznám v souřadnicích. Stačilo by, když by jsi vytvořl novou sestavu, která bude obsahovat pouze díly předního podvozku tak, aby střed osy podvozku procházel počátkem souřadnic.

Sestava aktuálně vypadá takto:
[image]

kaklik, 2019-11-13:
Proč je tu přední kolečko použitý jiný, než jsou zadní kolečka?  Reálně to jsou stejné modely..