ThunderFly-aerospace/TF-G1 #269: Změna parametrů nastavení předního kolečka

kaklik, 2019-11-13:
Parametry úhlů a závleku pro případ řízeného předního kolečka je potřeba upravit tak, aby osa zatáčení předního kolečka byla kolmá k vozovce a protínala osu rotace kolečka. 
Principielně tuto situaci vystihuje následující obrázek: 

[image]

Tlumiče, ale na kolečku zůstanou stejné jako jsou a jejich smyslem je tlumit vertikalní pohyb kolečka. 

Je potřeba zajistit, aby se tato změna promítla i do celkové sestavy.

kaklik, 2019-11-19:
Přední kolečko v plně nafouknutém stavu pístu nemá osu otáčení umístěnou tak, aby procházela s osou zatáčení. 

Kolečko při stlačení pístů může být až za osou otáčení - tedy vlečené. 

Preferovaný způsob řešení je změna délky prostředního dílu podvozku předního kolečka, neboť teď těsně zasahuje pod přední vrtuli a není tak možné na vírník namontovat větší průměr vrtule.

roman-dvorak, 2019-11-20:
V této sestavě (assembly_all_collisions.scad) je dobře vidět tento problém.

[image]

kaklik, 2020-02-13:
Mě se zdá, že v té sestavě chybí kulové igus ložisko na spodním pístu. 
Cože je součástí https://github.com/ThunderFly-aerospace/TF-G1/issues/201 v důsledku toho je ve vidlici kolečka zřejmě moc malý otvor na to, aby tam ložisko pasovalo. 
Nebo to má být otvor pro matku? 
Pak ale nevím, jak celá sestava předního kolečka má vypadat.

roman-dvorak, 2020-03-02:
Osa kolečka je stále trochu předsazena ose otáčení podvozku. Na fotce je vidět reálná délka pístu včetně igus ložiska a nasazeného dílu na pístu. To by bylo vhodné vykleslit do té sestavy.
Je to cca 138 mm. 

[image]