ThunderFly-aerospace/TF-G1 #270: Automatické slicování modelů přes makefile

kaklik, 2019-11-14:
Aktuálně slicer vypusuje tuto chybu: 

```
slicing/generic/basic_part.ini
   Slice STL/888_1016.stl with ./utils/slicer/PrusaSlicer-2.1.0.AppImage    slicing/888_1016.ini   
./utils/slicer/PrusaSlicer-2.1.0.AppImage: 1: ./utils/slicer/PrusaSlicer-2.1.0.AppImage: version: not found
./utils/slicer/PrusaSlicer-2.1.0.AppImage: 2: ./utils/slicer/PrusaSlicer-2.1.0.AppImage: oid: not found
size: '57700328': No such file
   Error in slicing
```

Je však potřeba ještě 

- [ ] Připravit unifikované nastavení sliceru pro většinu modelů
- [ ] Vytvořit specifické konfigurace pro některé díly

kaklik, 2019-11-24:
Problém se spuštěním spočíval v nenakonfigurovaném [git lfs](https://git-lfs.github.com/).

kaklik, 2019-11-25:
Vypadá to, že aktuálně generované g-kódy neobsahují inicializační sekvenci 

`;D3D:START:START
T0
M104 S215
G21
G90
M107
T0
G28
G29
M109 T0 S215
G0 F3000
M107
T0
G92 E0
G0 X0 Y149 Z0.1 F3600
G1 Y49 E6.5 F1200
G0 Z1
G92 E0
`

Navíc některé nepovnostní díly jako například lem krytu, mají více než jeden perimetr. Výplň dílů je také zřejmě typu zig-zag a není kubická.

roman-dvorak, 2019-11-25:
No, to se asi díváš na staré soubory. Protože nové (od cca poledne) to už obsahují. To samé platí i o infillu. Tento výchozí profil jsem přepsal z S3D profilu. Zřejmě bude potřeba udělat nějaké další úpravy konfigurace.

Co se týče dílů krytu, tak tam se předpokládá, že pro ně bude nějaký specifický profil sliceru.

kaklik, 2020-01-08:
Aktuálně slicování funguje.

closed 2020-01-08 by kaklik